djansen1987/SAJeSolar #1: Add Information eSolar SEC Module

LowKey88, 2021-07-27:
Hi,

I've eSolar SEC Module, can you add/capture data from eSolar SEC Module from fop.saj-electric.com?

This module can capture all this data as below :-
- Daily Load Energy
- Today Import Energy
- Total Pv Generation Energy
- Total Load Energy
- Total Import Energy
- Total Export Energy
- Total Pv Generation Consumption Energy
- Total Grid Power
- Total Grid Reactive Power
- Phase 1/2/3 Grid Voltage
- Phase 1/2/3 Grid Current
- Phase 1/2/3 Grid Frequency
- Phase 1/2/3 Grid Power
- Phase 1/2/3 Grid Power Factor
- Total Load Power
- Module Signal

This data is updated every 5 minutes on fop.saj-electric.com.

Much appreciate if you can included this information on this SAJeSolar.

Thanks

djansen1987, 2021-07-27:
Hi LowKey88,

Thank you for giving feedback and helping to make the module better. The configuration which i currently have to test with does not have the SEC module, there for I also do not have this data to process. But you can maybe help me process the data. I expect the data for this to be in the getPlantDetailInfo or getPlantMeterModuleList API part. 

For me to process this data you could send me the JSON output, i would advise you to send it in a private message as it contains Plant specific information you don't want to put on a forum. If decide to post it here please change/mask personal data like snList. To get this json output.

1. Login to the portal in preferable Chrome
2. Press F12 in chrome for the developer tools
3. Go to the "Network" tab
4. Enable "Preserve log"
5. Refresh the page
4. Filter on "getPlantDetailInfo"
5. Click on one of the results
6. Go to the response tab
7. Copy and share the complete response with me

An other option would be to create a viewer account to your portal which you trust me with to reverse engineer the API. But as we don't know each other i would fully understand this would not be an option for you.

--
It would also not surprise me if they have made a complete different api for this, if this is the case search for files that for example start with "get" to find other api endpoints. The api may also only appear in the developer tools by going to the right section on the page which displays the data you are looking for.

Api endpoints i have found until now:

- https://fop.saj-electric.com/saj/cloudMonitor/deviceInfo/getGridChartData
- https://fop.saj-electric.com/saj/monitor/alarm/getAlarmList
- https://fop.saj-electric.com/saj/monitor/site/getPlantDetailInfo
- https://fop.saj-electric.com/saj/monitor/site/getPlantDetailChart2?*********
- https://fop.saj-electric.com/saj/cloudmonitor/plantMeterModule/getPlantMeterModuleList
- https://fop.saj-electric.com/saj/cloudMonitor/device/findDevicePageList
- https://fop.saj-electric.com/saj/cloudmonitor/plantShare/getMyPlantShareList
- https://fop.saj-electric.com/saj/cloudmonitor/plantShare/getPlantVisitorsList
- https://fop.saj-electric.com/saj/monitor/site/getUserPlantList

--

[image]

LowKey88, 2021-07-28:
Hi djansen1987,

I just checked : https://fop.saj-electric.com/saj/cloudmonitor/plantMeterModule/getMeterModuleHistoryDataList

This will get data from eSolar SEC Modules.

Yes, I can share the account with you to investigate more about API's

How can I DM you?

djansen1987, 2021-07-29:
Hi LowKey88,

You can send it to github@djansen.nl if you like

LowKey88, 2021-07-29:
Hi @djansen1987,

I've sent the details :)

djansen1987, 2021-07-29:
Thank you, will let you know once i know more.

djansen1987, 2021-07-29:
> Hi djansen1987,
> 
> I just checked : https://fop.saj-electric.com/saj/cloudmonitor/plantMeterModule/getMeterModuleHistoryDataList
> 
> This will get data from eSolar SEC Modules.
> 
> Yes, I can share the account with you to investigate more about API's
> 
> How can I DM you?

On which page did you find this API url. I am unable to open the URL on your account by it self. The response i get says: "操作权限不足." what means as much as: "Insufficient operation authority". 

My guess is that there is send extra data along with the api such as an plant_id or cookie/session data.
When i can open the page with the login you provided (i changed the pwd) i can see what to send along to then get the right api output to work with.

LowKey88, 2021-07-30:
Hi Djansen,

Great work! :)

I've tested and it's working, but need some additional data from SEC Module as below :-

1.Total Load Power
2.Total Grid Power
3.Total PV Generation Power

This screenshot from eSolar SEC Module History Record and marked as red box as mentioned.
[image]

djansen1987, 2021-07-30:
Great to hear you like it. Coming 3 days i am busy and probably wont have time to work on it, when i have time again and worked on it i will post an update here.

djansen1987, 2021-08-08:
HI LowKey88,

I am unable to find the page you have send an screenshot of. It might have to do with te permissions i have.
Could you provide the navigation steps you have taken to get these details ?

LowKey88, 2021-08-08:
Hi Daniel,

Here the steps :- 

Go to Plant > Plant List > Operations Tab > Click on Green Pencil (Edit) > Load Monitoring >  Operations Tab > Click on Blue Graph Icon

You will see the tables data of Load Monitoring.

Let me know if this works

djansen1987, 2021-08-08:
> Hi Daniel,
> 
> Here the steps :-
> 
> Go to Plant > Plant List > Operations Tab > Click on Green Pencil (Edit) > Load Monitoring > Operations Tab > Click on Blue Graph Icon
> 
> You will see the tables data of Load Monitoring.
> 
> Let me know if this works

Hi LowKey88,

This what i was afraid of, with the current permissions i do not have access to that part. This is all i get:
[image]

ps: good news, i almost got the update ready with out of the box support for the Energy dashboard

LowKey88, 2021-08-08:
Hi Daniel,

I've changed your permission account. I think you know can see more options now. Try to repeat again the step.

Great about Energy Dashboard. I'm happy to see it soon! 👍

djansen1987, 2021-08-08:
HI LowKey88,

THanks, i am a bit further but don't see the blue graph icon. (i masked personal data)

[image]

djansen1987, 2021-08-13:
I think i have managed to add the sensors. I have not taken the approach of the api which was hidden. This was also only reachable with the correct permissions which looks only available in the powerview, which brings other issues. The data seemed also to be given via an other api, only in a large dataset. It was some puzzling but i think i managed to match the right data to the correct sensor, but it could also that i have mixed the sensors and i have to switch the dataset.

It was hard to get the right value as your plant was not generation at the moment of developing and there for the halve of the dataset was 0 which did make it almost impossible to get the right data.

Let me know if i need to switch the sensor names with the correct data. updated in V1.0.8

Again thanks for your support making the module better!

[https://github.com/djansen1987/SAJeSolar/releases/tag/v1.0.8](https://github.com/djansen1987/SAJeSolar/releases/tag/v1.0.8)

LowKey88, 2021-08-14:
[image]

[image]

Hi Daniel,

Great work!

So far it's work, except for sensor.esolar_totalgridpower . It's show zero value on most of the time. Can you recheck?

Thank you.

LowKey88, 2021-08-14:
[image]

current reading at this time is negative value, and maybe because of that it's shows 0.0 on HA.

djansen1987, 2021-08-14:
Hi thanks for testing. Could you check if the values are received after a restart of home assistant ? 
It seems connections to SAJ are dropped after some time. I thought this was because of expired cookies, but it still seems to happen.

LowKey88, 2021-08-15:
[image]

I just finished restart HA, and seem new value is accurate with the portal/apps SAJ. Seem like cookies is refreshed after restarting HA.

LowKey88, 2021-08-21:
[image]

Not getting any update 2 days ago. I did not restart my HA for last 2 days ago. Seem like this related to cookies?

djansen1987, 2021-08-23:
Hi,

There is an issue due to session or cookies. there is also an issue created for it https://github.com/djansen1987/SAJeSolar/issues/5
but it is mostly in dutch. Past week i was not able to work on the module, sorry. Hope to get time to look in to it again this week. But not sure yet on how to solve this. As all cookies are destroyed after each poll of data.

djansen1987, 2021-08-28:
https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.2