djansen1987/SAJeSolar #4: Integration with Energy Dashboard

LowKey88, 2021-08-04:
Hi @djansen1987 

Greetings,

Today Home Assistant release new update with new Energy Dashboard features

Here : https://www.home-assistant.io/blog/2021/08/04/home-energy-management/

Maybe you can integrate this to Energy Dashboard with SAJeSolar.

Thank you and have a nice day!

djansen1987, 2021-08-04:
HI @LowKey88,

I am looking in to it. As a workaround you can add the following to your configuration.yaml

```YAML
homeassistant:
  customize_glob:
    sensor.esolar_nowpower:
      device_class: energy
      state_class: measurement
      unit_of_measurement: kWh
      last_reset: "1970-01-01T00:00:00+00:00"
```

djansen1987, 2021-08-05:
extra info can be found here: https://github.com/cyberjunky/home-assistant-toon_smartmeter/issues/23

LowKey88, 2021-08-05:
```
homeassistant:
  customize_glob:
    sensor.esolar_nowpower:
      device_class: energy
      state_class: measurement
      unit_of_measurement: kWh
      last_reset: "1970-01-01T00:00:00+00:00"
```

for  sensor.esolar_nowpower , I think must use W instead of kWh

djansen1987, 2021-08-05:
> ```
> homeassistant:
>   customize_glob:
>     sensor.esolar_nowpower:
>       device_class: energy
>       state_class: measurement
>       unit_of_measurement: kWh
>       last_reset: "1970-01-01T00:00:00+00:00"
> ```
> 
> for sensor.esolar_nowpower , I think must use W instead of kWh

Your'e correct, but the HA energy module needs kWh. Therefor you can switch to the following sensor: sensor.esolar_todayelectricity

you can test with the following code:

```
homeassistant:
  customize_glob:
    sensor.esolar_todayelectricity:
      device_class: energy
      state_class: measurement
      unit_of_measurement: kWh
      last_reset: "1970-01-01T00:00:00+00:00"
```

I'm also working on the code to get the entities properly listed in HA but need some more time for that.

djansen1987, 2021-08-08:
Support has been added in V1.0.7

closed 2021-08-08 by djansen1987