djansen1987/SAJeSolar #5: esolar nowPower

megapeer, 2021-08-13:
De esolar nowPower blijft de volgende ochtend op 0 staan na herstart Home assistent werkt het weer.

in energie heb ik esolar todayElectricity  ingesteld bij de zonnepanelen 
De waarde is de volgende dag niet 0 en ik denk dat ie bij -10 of iets anders 
[image]
[image]
[image]
begint en gaat verder met optellen. Na herstart Home assistent telt de waarde af. Dit is alleen bij energie overzicht van Home Assistant

megapeer, 2021-08-13:
[image]

megapeer, 2021-08-13:
Dit overzicht is van gisteren 
[image]
[image]

djansen1987, 2021-08-13:
Hi @megapeer 

Het lijkt er inderdaad op dat de API na een dag stopt met werken. Ik ben aan het werken aan een fix. Ik zie geen error's op dat moment op de API wat vreemd is, maar mijn vermoeden is dat er een cookie verlopen is. Als tijdelijke oplossing zou je een automation kunnen maken die s'nachts HA herstart (begrijp dat dat niet wenselijk is). Ik zal me best doen dit zo snel mogelijk op te lossen.

djansen1987, 2021-08-13:
Should be fixed in V1.0.8: [https://github.com/djansen1987/SAJeSolar/releases/tag/v1.0.8](https://github.com/djansen1987/SAJeSolar/releases/tag/v1.0.8)

megapeer, 2021-08-14:
helaas werkt het niet,

moest ha toch restarten vannochtend.

megapeer, 2021-08-14:
Op Facebook zag ik iemand met zelfde probleem met andere omvormer, waarschijnlijk heb ik verkeerde sensor gekozen: esolar todayElectricity maar dat moet totaal zijn.

heb nu esolar totalElectricity ingesteld om te proberen en dat ziet er beter uit.

djansen1987, 2021-08-14:
Bedankt voor de terug koppeling. Het is volgens mij belangrijk dat alle sensors over dezelfde tijdsbestek zijn. Dus allemaal today, month, year of total

megapeer, 2021-08-15:
esolar nowPower en esolar todayElectricity blijven de volgende dag op 0 en moet ik ha herstarten en dan werkt het weer

ruudleijnse, 2021-08-19:
It is striking that it does detect that the unit is on again after sunrise. Only then will it not update the data.
Maybe it's an idea to use that item to trigger a reload of the data, or cookie cleaning?

djansen1987, 2021-08-23:
update: last week i was not able to work on the component. I will try to look in to it this week and hope to find an solution.

djansen1987, 2021-08-28:
V1.2.1 did not yet fixed it. After another hard look, i was passing the wrong variable to the API to provide the clientdate.

This is now fixed in:
https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.2

megapeer, 2021-08-28:
thanks, I am going to try

djansen1987, 2021-08-29:
Thanks, received already a positive feedback from one of the users. Let me know if it also works for you after 24 hours

megapeer, 2021-08-30:
thanks it works

djansen1987, 2021-08-30:
That is great news! Lots and lots of time have gone in to it, so glad to hear it is solved!

closed 2021-08-30 by djansen1987