djansen1987/SAJeSolar #6: eSolar SEC Module Issue

LowKey88, 2021-09-04:
Hi @djansen1987 

Greetings,

I'm still see issue on data provided for sensor.esolar_totalloadpower , sensor.esolar_totalgridpower & sensor.esolar_totalpvgenpower.

Here example data from HA
[image]

This from App SAJ

[image]

Sometime PV generation data seem swapped with Grid output data. Not sure why this happen, can you look up on this?

Thank you.

djansen1987, 2021-09-06:
Hi @LowKey88 ,

Could you see if it is fixed in the latest update ? Is switched the parameters. At creating i was not 100% sure which was which so i might have mixed the up.

djansen1987, 2021-09-06:
```
            if self._type == 'totalGridPower':
                if 'dataCountList' in energy:
                    if energy['dataCountList'][4][-1] is not None:
                        self._state = float(energy['dataCountList'][3][-1])
            if self._type == 'totalLoadPower':
                if 'dataCountList' in energy:
                    if energy['dataCountList'][4][-1] is not None:
                        self._state = float(energy['dataCountList'][2][-1])
            if self._type == 'totalPvgenPower':
                if 'dataCountList' in energy:
                    if energy['dataCountList'][4][-1] is not None:
                        self._state = float(energy['dataCountList'][4][-1])
```

This is the part there i get the data out of the api. As it is not properly labeled, and it is also an array of data i had to work with positions instead of labels. I cannot imagine they are switching the positions of this data but if they are then it might become very hard to get this right, as i only found 1 more API endpoint which is only accessible via the reseller/main login which i believe cannot be used in the component because of various reasons..

djansen1987, 2021-09-06:
Found an new API endpoint which if it has the correct info is way easier to understand for me. Could you see the bold items are the values you are looking for ? Found it the data under load monitoring - >Overview. it represents the data shown in the image below :

[image]

https://fop.saj-electric.com/saj/monitor/site/getPlantMeterEnergyPreviewInfo?plantuid=1234-aa-12346-fd123-123423-&moduleSn=M1234Y12345678&_=1830669656746

previewInfo: 
gridDirection: -1
-> **gridPower**: 1723
isOnline: 1
moduleSn: "M1234Y12345678"
-> **pvPower**: 0
solarPower: 12
-> **totalLoadPower**: 1723
status: "success"

djansen1987, 2021-09-08:
Hi @LowKey88 

Should be fixed in [v1.2.4](https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.4).

Please read and follow up on the instructions within the release notes. As per my findings we were on the wrong track of getting the correct sensors to make the Energy Dashboard work as it should. There for other sensors were needed (see release notes)

Could you confirm if my findings also work for you? Also if it does, we might be able te deprecate some old sensors which are not needed any more, this can make the component faster en cleaner. I think the following sensors are not needed anymore after the latest update, could you confirm?

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate

      - totalGridPower
      - totalLoadPower
      - totalPvgenPower

djansen1987, 2021-09-17:
Fixed with above solutions

closed 2021-09-17 by djansen1987