djansen1987/SAJeSolar #7: total_increasing

djansen1987, 2021-09-06:
[image]
https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

djansen1987, 2021-09-06:
Should be fixed in latest update

LowKey88, 2021-09-06:
Hi @djansen1987 

Still getting same issue after updated. 

[image]

djansen1987, 2021-09-08:
Should be fixed in [v1.2.4](https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.4). Please see release notes for details.
Please let me know if this solves the Issue.

closed 2021-09-08 by djansen1987