djansen1987/SAJeSolar #8: unit_of_measurement

megapeer, 2021-09-07:
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:151
Integration: Sensor (documentation, issues)
First occurred: 21:58:16 (15 occurrences)
Last logged: 21:58:19

custom_components.saj_esolar.sensor is setting 'unit_of_measurement' on an instance of SensorEntityDescription, this is not valid and will be unsupported from Home Assistant 2021.11. Please report it to the custom component author.

djansen1987, 2021-09-07:
Thanks for reporting! Appreciated. Will need some time for this as it will resolve in a lot of programming and again restructuring of the component (again).

djansen1987, 2021-09-08:
Should be fixed in [v1.2.4](https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.4). Please see release notes for details.
Please let me know if this solves the Issue.

djansen1987, 2021-09-17:
Fixed with above solutions

closed 2021-09-17 by djansen1987

bertdebondt, 2021-12-07:
bug "custom_components.saj_esolar.sensor is setting 'unit_of_measurement' on an instance of SensorEntityDescription, this is not valid and will be unsupported from Home Assistant 2021.11. Please report it to the custom component author." is reappearing in my log file.

djansen1987, 2021-12-07:
Hi @8OND007, which version of saj_esolar are you running. This should be fixed in a couple of version back.

8OND007, 2021-12-07:
Version 1.2.4 according to the manifest.json file.

Strange HACS did not see 2 updates 1.2.5 and 1.2.6.
I just used the "Redownload" function in HACS so I though I would have 1.2.6.
I restarted HA with the new 1.2.6 but in log I still see : "custom_components.saj_esolar.sensor is setting 'unit_of_measurement' on an instance of SensorEntityDescription, this is not valid and will be unsupported from Home Assistant 2021.11. Please report it to the custom component author."
Again checked manifest.json, and it is still 1.2.4.
Maybe something is wrong with the update version push ?

8OND007, 2021-12-07:
the actual manifest.json file on github is still on 1.2.4.
This is probably the cause my HACS does not see a newer update ?

djansen1987, 2021-12-07:
Hi, i updated the manifest.json, could you check again

djansen1987, 2021-12-07:
Continue issue in https://github.com/djansen1987/SAJeSolar/issues/19