djansen1987/SAJeSolar #16: AS1-3KS-5.1 (Untested)

BeardyBrah, 2021-11-26:
Hello @djansen1987, 
Thank you for your amazing work! I've got the untested SAJ AS1-3KS-5.1 5,1kWh battery (I beleive these got rebranded as an Eveready Energy Vault) and would be happy to contribute in any way I can to this project.

Its paired with a R5-5K-S2 Inverter and (from what I can tell) the WiFi-D data collector.

So far I've got the saj_esolar portion of your project working great, but can't seem to pull anything from the Battery or self consume rates. Happy to test/assist just let me know

BeardyBrah, 2021-11-27:
Adding notes;
When logged into the eSolar Portal and on the Battery Real Time Data page, Checking Inspect Element > Network > findBatteryRealTimeList > Response I've found the following that might be of use (formatted for my yaml, but its currently invalid)

`     - batVoltage
      - batSOC
      - batCapcity
      - batCurrent
      - batPower
      - batTemperature`

BeardyBrah, 2021-11-27:
Diffrent issue (probably me)
In the Energy section of HA, when setting a Solar Panel I assume I'd be setting it as the 'eSolar nowPower' for live incoming production. It doesnt appear in the drop-down list, so would this be due to a state_class ?

I've tested instead with 'esolar totalElectricity' but it's WAY too high (9.8 kWh) to be whats actually on my roof (5.5).
[image]

djansen1987, 2021-11-28:
Hi Beardy, Thanks for your info. Could you try to use the "H1" code in the manual ? The H1 has also a Battery.

```yaml
    sensors: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower
```

BeardyBrah, 2021-11-28:
Sure I'll give it a try. Whenever I try to use it however I get;

`Integration error: sensors - Integration 'sensors' not found.
Integration error: resources - Integration 'resources' not found.`

Heres what my configuration.yaml looks like

`#SAJ eSolar HACS Integration
sensor:
  - platform: saj_esolar
    username: !secret saj_username
    password: !secret saj_password
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # TOTAL Energy -> Solar production
      - todayGridIncome #comes up as unknown
      - income #comes up as unknown
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower #comes up as unknown
#SAJ V2 Card (just testing more available sensors)
      - nowPower
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity
      - totalConsumpElec
      - totalBuyElec #comes up as unknown
      - totalSellElec
      - selfUseRate
#SAJ V3 Card (removing from lovelace as nothing works)
      - pvElec #comes up as unknown
      - useElec #comes up as unknown
      - buyElec #comes up as unknown
      - sellElec #comes up as unknown
      - buyRate #comes up as unknown
      - sellRate #comes up as unknown
      - totalGridPower #comes up as unknown
      - totalLoadPower #comes up as unknown
      - totalPvgenPower #comes up as unknown
sensors: h1
resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower`

djansen1987, 2021-11-29:
Hi, please only add the sensors which belong to the type.
The unknown entities can be deleted within Home Assistant

The full config for you would be:

```yaml
#SAJ eSolar HACS Integration
sensor:
  - platform: saj_esolar
    username: !secret saj_username
    password: !secret saj_password
    sensors: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower
```

BeardyBrah, 2021-11-29:
Thank you. Ahh, I now see how that works.
configuration.yaml edited, checked (OK) and rebooted.

Most of the sensors work, and are available for me to throw them onto an Entity Card except;

```
      - batteryPower #this entity could not be found
      - gridDirection #this entity could not be found
      - outPower #comes up as unknown (could be due to its night time here)
      - pvDirection #this sensor could not be found
```

djansen1987, 2021-11-29:
Hi @BeardyBrah,

That is great to hear. Probably those sensors are not available for your unit of named differently. You can remove them from the resources list and delete the unknown entities in HA. Are there any sensors that you are missing now ?

BeardyBrah, 2021-11-29:
I'm trauling though looking for anything missing. On the SAJ home page there is a section that reads the current Battery output

[image]

and came across this section that I couldn't seem to match up a sensors for

[image]

LoadPower - this is the power in Watts the battery is currently putting out, basically the same as screenshot 1.
Battery Voltage/Current/Power
Battery Total Charge/Total Discharge

BeardyBrah, 2021-11-29:
Unrelated, renaming sensors to better understand what they are. Found a couple of items that missed having a suffix after the value ;)

sensor.esolar_totalconsumpelec - kWh
sensor.esolar_batcapcity - Ah
sensor.esolar_batenergypercent - %
sensor.esolar_gridpower - W

[image]

djansen1987, 2021-11-29:
> I'm trauling though looking for anything missing. On the SAJ home page there is a section that reads the current Battery output
>
> [image]
>
> and came across this section that I couldn't seem to match [...]

Could you find out in the developer tools under network which api it uses and then send me the api structure (remember not to post personal detail, only need the api url and the api structure). You can sort the network tab by fetch. (reload of the page is needed when the developer tools was not open when loading the details.

[image]

BeardyBrah, 2021-11-29:
Not sure exactly where to find which API it uses, or its File Structure, but heres what I've got (Personal info replaced with 'REDACTED');

https://fop.saj-electric.com/saj?lang=en

> findDevicePageList

`{"result":"OK","page":"<ul>\n<li class=\"disabled\"><a href=\"javascript:;\">« Previous</a></li>\n<li class=\"active\"><a href=\"javascript:;\">1</a></li>\n<li class=\"disabled\"><a href=\"javascript:;\">Next »</a></li>\n<li class=\"disabled controls pageC\">Page <input type=\"text\" value=\"1\" onkeypress=\"var e=window.event||event;var c=e.keyCode||e.which;if(c==13)findDevicePageList(this.value,10,'');\" onclick=\"this.select();\"/> Total 2 Records</ul>\n<div style=\"clear:both;\"></div>","list":[{"kitType":"N/A","monthSellEnergyStr":"749.56","todaySellEnergy":38.06,"kitSn":"REDACTED","updateDateStr":"2021-11-29 17:32:50","powernower":"N/A","mastermcufw":"V1.171","type":0,"devicetype":"R5-5K-S2","onLineStr":"3","displayfw":"V2.061","devicepc":"REDACTED","powernow":0.0,"owner":"N/A","todaySellEnergyStr":"38.06","index":0,"monthSellEnergy":749.56,"devicesn":"REDACTED","userId":"REDACTED","onLine":0,"slavemcufw":"V1.171","isModuleExpire":0,"totalSellEnergy":2977.48,"totalSellEnergyStr":"2,977.48","isHistory":0,"mark":1,"plantName":"REDACTED","dataTimeStr":"2021-11-29 19:32:00"},{"kitType":"N/A","monthSellEnergyStr":"742.59","todaySellEnergy":37.29,"kitSn":"N/A","updateDateStr":"2021-11-29 19:42:32","powernower":"2","mastermcufw":"v5.031","type":2,"devicetype":"REDACTED","onLineStr":"1","displayfw":"v3.094","devicepc":"REDACTED","powernow":2.0,"owner":"N/A","todaySellEnergyStr":"37.29","index":0,"monthSellEnergy":742.59,"devicesn":"REDACTED","userId":"REDACTED","onLine":0,"slavemcufw":"v5.031","isModuleExpire":0,"totalSellEnergy":2970.34,"totalSellEnergyStr":"2,970.34","isHistory":0,"mark":1,"plantName":"REDACTED","dataTimeStr":"2021-11-29 21:41:30"}]}`

> getGridChartDate isnt available, but I do have instead;

> getPlantDetailChart2?plantuid..........................
>getStoreOrAcDevicePowerInfo?plantuid=.......................

djansen1987, 2021-11-29:
think you are almost there. Quick look from my side i think you are looking for one of these 2:

url: https://fop.saj-electric.com/saj/monitor/site/getStoreOrAcDevicePowerInfo?plantuid=&devicesn=XXXXXXXXXXX&_=XXXXXXX
API Structure:
```yaml
status: "success"
storeDevicePower: 
  batCapcity: xxx
  batCurr: 0
  batEnergyPercent: xx
  batteryDirection: 0
  batteryPower: 0 # <----- this one
  dataTime: xxxx
  gridDirection: -1
  gridPower: xx
  isAlarm: 0
  isOnline: 1
  mark: 1
  outPower: xxx # <----- or this one
  outPutDirection: 1
  pvDirection: 1
  pvPower: xxx
  runningState: 1
  solarPower: 0
  totalLoadPower: xxx
  type: 1
  updateDate: xxx
  type: 1
```

BeardyBrah, 2021-11-29:
batteryPower is the one
I can't see an 'outPower' anywhere.

djansen1987, 2021-11-30:
Hi @BeardyBrah ,

First of al many thanks on the donations! Much appreciated! 
Second i think all should be resolved in the new release. Please let me know if this has solved it for you.

release v1.2.6 https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.6

djansen1987, 2021-12-01:
Solved

closed 2021-12-01 by djansen1987