djansen1987/SAJeSolar #18: Current House Load Power

BeardyBrah, 2021-12-01:
Hi again @djansen1987,
Hoping to continue assisting with anything I'm finding from my end. I've noted I can't see a 'Current Home Load/Power' statistic to read how much the home is using throughout the day.

[image]

a few unrelated finds;
sensor.esolar_solarpower should also have the suffix kWp.

sensor.esolar_gridpower doesn't seem to know if it's importing power or exporting power (in Watts). In either direction it reads as a positive number either way.

sensor.esolar_batterydirection has a number as a value, I've found -1 is Charging from PV, 1 means Discharging to home, 0 means standby/not currently being used. Can the value it outputs be changed to 'CHARGING / DISCHARGING / STANDBY' ?

Thanks mate! Love your work

BeardyBrah, 2021-12-01:
Extra info that may help.

https://fop.saj-electric.com/saj/monitor/site/detailForm?plantuid=xxxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx

```
pvPower: 54.00 #PV being generated
gridPower: 63.00 #feeding to grid realtime
inputOutputPower: 546.00 #dont know what this is
batteryPower: 1006.00 #battery discharging in W
totalLoadPower: 997.00 #real time home power usage
homeLoadPower: 550.00 #dont know what this is
backupLoadPower: 447.00 #dont know what this is
batCurr:19.50 #dont know what this is
batEnergyPercent: 28.00 #battery state of charge
pvDirection:1 #currently outputting
gridDirection:1 #currently exporting to grid
batteryDirection:1 #currently discharging
outPutDirection:1 #dont know what this is
```

[image]

BeardyBrah, 2021-12-23:
Hey @djansen1987 
Can I be of any further assistance on this one? Need any extra information?

djansen1987, 2021-12-25:
Hi @BeardyBrah, december has been a busy month. Hope to find time in januari to take a look at it

djansen1987, 2022-01-10:
Hi @BeardyBrah,

Thanks for the enhancement and details below the answers:

>  I've noted I can't see a 'Current Home Load/Power' statistic to read how much the home is using throughout the day.

it was not added to the H1 module and is added in the upcoming update, add resource: totalLoadPower

> a few unrelated finds;
> sensor.esolar_solarpower should also have the suffix kWp.

For what i can find this is in Watt not in kWp

> sensor.esolar_gridpower doesn't seem to know if it's importing power or exporting power (in Watts). In either direction it reads as a positive number either way.

this can be known by the following sensor: outPutDirection
If i am correct in the esolar portal the number is also always positive. I understand your thoughts, for now not planning to change this.

> sensor.esolar_batterydirection has a number as a value, I've found -1 is Charging from PV, 1 means Discharging to home, 0 means standby/not currently being used. Can the value it outputs be changed to 'CHARGING / DISCHARGING / STANDBY' ?

Changed this in upcoming update 1.2.8

djansen1987, 2022-01-14:
Hi @BeardyBrah

I was wondering if it is working for you and if i can close the issue ?

BeardyBrah, 2022-01-14:
Hi mate,
Just saw this now. Heres what mine displays

[image]

BeardyBrah, 2022-01-14:
Updated my configuration.yaml to include totalLoadPower, fixed. Looking good

[image]

djansen1987, 2022-01-15:
Great! Thanks for pointing out! Have a great weekend!

closed 2022-01-15 by djansen1987