djansen1987/SAJeSolar #21: Home Battery Storage (Energy Integration)

BeardyBrah, 2022-01-14:
Hi Mate,
Thank you for the new updates. Your work is incredible. Happy to keep assisting in any way I can.

The final enhancement could be under the Energy Dashboard. The Home Battery Storage section allows to monitor battery Input & Output. 

[image]

Output via "sensor.esolar_batterypower" can't be selected.
I don't beleive we've found a method to read the input (yet...) I'll do some research and see if I can find something and report back

djansen1987, 2022-01-15:
i think it must be these to:
[image]

I then need to know the parameters in the API and then find out which unit of measurement home assistant wants these in

djansen1987, 2022-01-15:
Think these are the ones:
https://fop.saj-electric.com/saj/monitor/site/getPlantDetailChart2?plantuid=xxxxxxx
viewBean ->chargeElec
viewBean ->dischargeElec

djansen1987, 2022-01-15:
Found them, created new release. Please let me know if it works for you ?

https://github.com/djansen1987/SAJeSolar/releases/tag/v1.2.9

BeardyBrah, 2022-01-15:
Its reading something, however I don't think its the right value. 
Heres my SAJ Portal vs. HA readings
[image]
[image]

djansen1987, 2022-01-15:
Are you able to look at the api level to see the values ? Could be Convert from watt to kwp over a longer term

djansen1987, 2022-01-15:
Also look at the inverter view,there you can find the screenshot I taken. 

BTW energie dashboard needs kwp not watt if an correct for battery

BeardyBrah, 2022-01-15:
few things I've found 

sensor.esolar_batterypower appears to read both the input (Charging) and the output (discharging) of the battery and can't dissern between the two. 

Same appears true for gridPower is the power im sending to the grid. This reflects sensor.esolar_gridpower but I have a feeling it reads both both input to the grid and consumption. Will report back on this one

Is it possible to code something along the lines of
IF batteryDirection = -1
THEN batteryPower = Charging
OUTPUT 1113 to esolar chargeElec (or a purpose built sensor)

IF batteryDirection = 1
THEN batteryPower = Discharging
OUTPUT x_amount to esolar dischargeElec (or again another sensor to read this value)

Same for Grid power I'd assume (if any of this makes any sense lol...I don't code)

[image]
.

BeardyBrah, 2022-01-16:
I've taken a look at the end of the day - I've highlighted everything that matches however can't seem to verify the daily of battery in/out.

[image]

djansen1987, 2022-01-17:
1. what if you go to: (inverter)
[image]

2. then click on the module number:
[image]

3. scroll down to the bottom of the page:
[image]

BeardyBrah, 2022-01-17:
The statistic shows me an overall usage by the looks of it
Doesnt break it down over a 24 hour use

[image]

Having said that, the sensors you've found appear to be reading correctly (at a guess), heres todays usage.

14.9 kWh consumption in the home (total)
around 50% from solar directly
20 %'ish from the battery
30 %'ish from grid imports

[image]

djansen1987, 2022-01-17:
The home assistant Energy Dashboard takes the total, as this number should only increase it can there for know the usage over time by calculating it's first value and date/time and so on. Thanks for confirming!

closed 2022-01-17 by djansen1987