djansen1987/SAJeSolar #22: "return data NONE" occurs many time recurrent in HA log

8OND007, 2022-01-18:
Home Assistant log get's filled with 100's, 1000's error occurences :

see event below which occured 108 times on maybe 1 or 2 minutes : 

 Logger: custom_components.saj_esolar.sensor
Source: custom_components/saj_esolar/sensor.py:797
Integration: saj_esolar (documentation, issues)
First occurred: 20:46:39 (108 occurrences)
Last logged: 20:49:09
return data NONE

also got the following 1 time :  

 Logger: custom_components.saj_esolar.sensor
Source: custom_components/saj_esolar/sensor.py:768
Integration: saj_esolar (documentation, issues)
First occurred: 20:48:39 (1 occurrences)
Last logged: 20:48:39
Unknown error occurred while polling eSolar: Session is closed

I just activated the debug log on HA and will check what it says over there.

djansen1987, 2022-01-19:
Hi @8OND007 ,
This must mean there is a authentication or configuration error in your setup and debug logging is enabled. Could you share your esolar configuration (without username password) and what device module you are trying to setup ?

8OND007, 2022-01-19:
- platform: saj_esolar
    username: my secret username
    password: my secret password
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - todayGridIncome
      - income
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower

SAJ Sun Uno Plus 5K-M  (5 kW) (connected through a Wifi Dongle on the inverter)

looks like this:
[image]

data in lovelace: 

[image]

I fully restarted HA with debug log activated on your integration now and I don't seem to be getting the error anymore. Strange, will keep my eye on it.

I also use my own fork of the official SAJ integration (local IP access) which I modded for my type of inverter.
The original SAJ HA integration does not support or add AC / DC Voltages/Currents/Cos Phi, ... which is available though in the inverter. So I changed the original parser code for my type of inverter.
Problem with official integration also is during night values are not available which causes the total counter and HA new Energy dashboard to go out of range and start the day with +30xxxx kWh production.
That's the reason I use your cloud SAJ Integration since the counter in the cloud stays at last known values.
I don't see AC & DC voltage and currents however in your integration. It is also available on the cloud SAJ website though. Maybe you can add them also on your integration in future ?

On SAJ website : inverter info :  below : 

[image]

also on the realtime page : 

[image]

djansen1987, 2022-03-03:
Hi @8OND007 working with the data from the realtime page is a nightmare in the api. Your first screenshot i will have to lookup to see in witch api they hide this information. 

What would it be from this first screenshot you would like to have in Home assistant.
Maybe you can make the concept sensors like so:

sensor.AC1Electricity
sensor.AC1Voltage
sensor.AC1Frequency

sensor.AC2Electricity
sensor.AC2Voltage
sensor.AC2Frequency

sensor.AC3Electricity
sensor.AC3Voltage
sensor.AC3Frequency

sensor.DC1Electricity
sensor.DC1Voltage
sensor.DC1Frequency

sensor.DC2Electricity
sensor.DC2Voltage
sensor.DC2Frequency

sensor.DC3Electricity
sensor.DC3Voltage
sensor.DC3Frequency

closed 2022-05-06 by djansen1987