djansen1987/SAJeSolar #41: what is totalloadpower

santiagozky, 2022-11-03:
Hello,
First of all, thank you for this. it's been very helpful!

Im having trouble understanding what the sensor _totalloadpower_ is. looking at the SAJ portal it seem to indicate the power being drawn by my home (from getPlantMeterEnergyPreviewInfo ) but in HA this seems to be reporting only the self consumed load. e.g. If Im loading 1kw from the grid and 1kw from my panels, I was expecting totalloadpower to report 2kw, but it just reports 1kw.

Is this a bug or or Im just not understanding it's purpose. Which sensor contains the total load being drawn by my home?

djansen1987, 2022-11-06:
Hi @santiagozky, good question. I do not own a H1 of SEC module and cannot really tell you to be honest.

Maybe @santiagozky , @faanskit @robshot can answer this question?

robshot, 2022-11-06:
I will write it down to check it out tomorrow.

robshot, 2022-11-07:
@santiagozky I'm looking at my totalloadpower and that is the total grid power + total solar power. So I'm currently having 446W of solar power with 552W of the grid and my total load power is 998W.
I don't know why on your side it doesn't show the 2kW but that is indeed what it should show and how it's working on my setup.

santiagozky, 2022-11-07:
thank you for clarification very much.  It then seems to be something on my side.

santiagozky, 2022-11-10:
hello again!
I was debugging this and it took me a while (Im new to HASS and Python). I think my setup might be different from yours, because in my case it seems the totalLoadPower is coming in a different column in getPlantMeterDetailInfo.

Checking my requests and comparing to my graph in https://fop.saj-electric.com/saj, I see that totalLoadPower matches the second column, not the third. the columns I get are:

0- PV power 
1- load power
2- self consumption
3- export power
4- import power

which matches the order in my UI :
[image]

Looking at the code it seems those don't match to the code expectations.

What could be the cause of this? I dont have an H1, I have a R5 (rebranded as Greenheiss) with a sec module.

robshot, 2022-11-15:
I have the same order as you in my web interface, so I don't think that order affects how we receive the data.
[image]
I'm not sure why the data is not in the same order, I would suspect that the rebrand doens't matter either.

Currently I'm very busy but I might look into rewriting a part of the code to also support the realtime info.
Don't expect the changes soon probably even only next year.

santiagozky, 2022-11-15:
I dont think the rebrand is the issue either. I access the SAJ branded portal (the rebranded portal is http://inversores-style.greenheiss.com which just has a different color).
I changed the code locally to feed from column 1 instead of 2 and now I get the right readings. I could open a PR, but if Im the only one with this issue I suppose there is something more in play.

robshot, 2022-11-15:
Do you have a battery system?
Else somebody is creating a new integration which could be working for you but there is no battery support for now.
https://github.com/djansen1987/SAJeSolar/discussions/45#

santiagozky, 2022-11-15:
I dont have a Battery. for the time being Im happy leaving things as they are. I have a private fork just adjusting the columns as I receive them and it works well so far. but would be nice if it was clarified if there is a difference in data between battery and non-battery systems to better improve this.

robshot, 2022-11-15:
normally there is no differences as I have a battery setup but other people that are helping here which don't have this and it's also working.

santiagozky, 2022-11-16:
Just to be clear. this is what I changed locally in sensor.py after inspecting the contents of `https://fop.saj-electric.com/saj/monitor/site/getPlantMeterChartData`

```
--- a/custom_components/saj_esolar/sensor.py
+++ b/custom_components/saj_esolar/sensor.py
@@ -1158,15 +1158,15 @@ class SAJeSolarMeterSensor(SensorEntity):
                 if self._type == 'totalGridPower':
                     if 'dataCountList' in energy:
                         if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
-                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][3][-1])
+                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][4][-1])
                 if self._type == 'totalLoadPower':
                     if 'dataCountList' in energy:
-                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
-                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][2][-1])
+                        if energy["getPlantMeterChartData"]['dataCountList'][1][-1] is not None:
+                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][1][-1])
                 if self._type == 'totalPvgenPower':
                     if 'dataCountList' in energy:
-                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
-                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][4][-1])
+                        if energy["getPlantMeterChartData"]['dataCountList'][0][-1] is not None:
+                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][0][-1])
 
 
```

faanskit, 2022-11-16:
I do not have an answer to this, but looking into the plantChart2.js code, I take note of the following different titles:
```
var titleArray = [PV_POWER,LOAD_POWER,SELF_POWER,SELL_POWER,BUY_POWER]
var titleArray = [PV_POWER,LOAD_POWER,SELF_POWER,SELL_POWER,BUY_POWER,BAT_POWER,"SOC"}
var titleArray = [PV_PRODUCTION,CONSUMPTION,SELF_ENERGY,EXPORT_ENERGY,IMPORT_ENERGY]
```
The behavior seems to differ depending on what is sent as input in `chartDateType` in the `/getPlantDetailChart2` endpoint.
E.g.
```
if(chartType=="2"){//发电功率/发电量
if(chartType=="3"){//发电效率
```

- 发电功率/发电量 translates to "generating power/power generation"
- 发电效率 translates to "power generation efficiency"

So, it *appears* as if the request may differ for different systems. I think the way forward is to capture (in Chrome) how the request looks for @santiagozky compared to other systems.

In particular, the `/getPlantDetailChart2` is what needs to be traced and compared, I think.

I *think* the key to this is hidden in the respons parameter pvType:
[image]

Again, the clues are in the plantChart2.js code:
```
if (data.pvType == 12) {
  for ( var k = 0; k < dataCountList.length; k++) {

} else if (data.pvType == 4) {
  for ( var k = 0; k < dataCountList.length; k++) {

if (data.pvType == 3) {
  for ( var k = 0; k < pvDataNameArr.length; k++) {
    if ((k >= 4 && k <= 12) || (k >= 16 && k <= 24)
      || (k >= 28 && k <= 36)) {

} else if (data.pvType == 4) {
  for ( var k = 0; k < pvDataNameArr.length; k++) {
    if ((k >= 5 && k <= 12) || (k >= 17 && k <= 24)
      || (k >= 29 && k <= 36)) {
```

So, by looking at the pvType parameter from a few different system, this is an easy fix.

HTH/Marcus

santiagozky, 2022-11-16:
the input `chartDateType`  is 1 for "day" since it's graph is different than "month" and "year" which are graph bars. if you switch from today to month you'll see a second request with chartDateType.

Honestly, it seems to me like this is a bug, but appears only if you don't use the h1 sensor.

If you defined the H1 sensor the totalLoadPower is taken from sensor.py line 1100:

``` if self._type == 'totalLoadPower':
                    if 'totalLoadPower' in energy["storeDevicePower"]:
                        if energy["storeDevicePower"]['totalLoadPower'] is not None:
                            self._state = float(energy["storeDevicePower"]['totalLoadPower'])
```

It seems that if you have a battery, that data comes from the storeDevicePower object.

If you use the saj_sec sensor, it comes from line 1165:

```  # dataCountList
                if self._type == 'totalGridPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][3][-1])
                if self._type == 'totalLoadPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][2][-1])
                if self._type == 'totalPvgenPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][4][-1])
```

this comes from request to `/getPlantMeterChartData`. You mentioned  `/getPlantDetailChart2` but that seems to be for the photovoltaic generation only (the "home" tab), not energy monitoring ("load monitoring" tab).

Looking at `/getPlantMeterChartData`, when using  `chartDateType=1`. it has a dataCountList object with columns for this titles `[PV_POWER,LOAD_POWER,SELF_POWER,SELL_POWER,BUY_POWER];` (line 347 of plantMeterChart.js).

But again, looking at the code above, it expects 

- totalGridPower in ['dataCountList'][3] (but should be 4)
- totalLoadPower in ['dataCountList'][2] (but shoud be 1)
- totalPvgenPower in ['dataCountList'][4] (0)

faanskit, 2022-11-17:
> the input chartDateType is 1 for "day" since it's graph is different than "month" and "year" which are graph bars. if you switch from today to month you'll see a second request with chartDateType.

Ok, makes sense.

> @faanskit do you use h1 or saj_sec sensor?

Neither, I only have a R5 system without SEC.

> Honestly, it seems to me like this is a bug, but appears only if you don't use the h1 sensor.

Bug or feature. What I try to [help you] figure out is if there is a condition that can be used to report differently for different systems. From the discussion, both appear to be correct - but depending on configuration (what system and modules you have)

It appears as if this integration considers H1 and SEC as mutually exclusive. I would assume you can have a SEC installed also for a H1 system.
`        vol.Optional(CONF_SENSORS, default="None"): cv.string,
`
What happens if you configure both? My _assumption_ is that one will be overwritten, which can be the source of this bug/feature.

What happens if you change order of them in your `configuration.yaml`?

I think the root cause it that H1 is considered to be a module, when in fact it is a Hybrid Inverter. Sure, since H1 has a battery it has additional measurement points than e.g. the R5 series.

The SEC however should be considered an optional module, that works both with R5 and H1 systems.

I do not have a solution, and honestly I don't even have the relevant systems. But I gladly bounce ideas.

closed 2022-12-17 by djansen1987