djansen1987/SAJeSolar #46: Battery temperature sensor

cypherbits, 2022-12-10:
H1 with batteries you can get the average battery temperature.

Will make a pull request soon. I have this working.

cypherbits, 2022-12-13:
@djansen1987 Some questions about the lastest beta:

- why did you remove the sensors conditions (the if h1, or sec)? 
- why did you remove the sensor list from configuration?
- my HA says the "version": "1.4.1.B1" is not a good version number.

I implemented the batteryTemperature that is working for me on a H1.
Reenabled the configuration and the sensors check.
I want to check for the configuration batteryTemperature before calling and trying to get the temperature, that is why I reenabled them.

djansen1987, 2023-01-25:
Hi @cypherbits,

THanks for your interest, I am looking in to making the setup more easy. No more advanced config yaml, and let the component find out what type or modules you have. But as you can see i am a bit stuck. This is partly due to the lack of enough python knowledge and also time. 

Maybe you can help out, only thing important to me is backward competible sensor names to maintain Energy dashboard statistics.

djansen1987, 2023-03-10:
I have checked one of my account with a battery but can't seem to find a api endpoint showing battery tempature. Please reopen the issue if you have found more information.

closed 2023-03-10 by djansen1987

cypherbits, 2023-03-13:
Some code I have:
```
url_findBatteryRealTimeList = f"https://fop.saj-electric.com/saj/cloudMonitor/deviceInfo/findBatteryRealTimeList"
                #"2022-11-10+21:30:00"
                datetime_now = datetime.datetime.now().strftime("%Y-%m-%d+%H:%M:%S")
                _LOGGER.debug(datetime_now)
                payloadBattery= f"devicesn={deviceSnArr}&timeStr={datetime_now}"
                response_findBatteryRealTimeList = await self._session.post(url_findBatteryRealTimeList, headers=headers, data=payloadBattery)
                result_findBatteryRealTimeList = await response_findBatteryRealTimeList.json()
                result_batteryTemperature = dict()
                result_batteryTemperature["batteryTemperature"]=result_findBatteryRealTimeList["list"][0][0]["batTemperature"]
                plantDetails.update(result_batteryTemperature)
                _LOGGER.debug(result_findBatteryRealTimeList)

                if response_findBatteryRealTimeList.status != 200:
                    _LOGGER.error(f"{response_findBatteryRealTimeList.url} returned {response_findBatteryRealTimeList.status}")
                    return
```