djansen1987/SAJeSolar #50: Integration is using deprecated `DEVICE_CLASS_*` constants

frenck, 2023-01-03:
This custom integration uses deprecated `DEVICE_CLASS_*` constants in its codebase.

The `DEVICE_CLASS_*` constants have been deprecated and replaced in Home Assistant Core 2021.12 (over a year ago). I would highly suggest updating/migrating this integration to the new enums.

For example, for the device classes supported by the `sensor` platform, there is now a `SensorDeviceClass` enum. So if a sensor previously used the `DEVICE_CLASS_ENERGY` constant, it should now use `SensorDeviceClass.ENERGY`. Other platforms (like `binary_sensor`, and `number`) provide similar enumerations for their supported device classes.

The migration thus only consists of replacing constants with an enumeration member and is, therefore, very low impact and should be fairly straightforward.

If I can help resolve any questions regarding this change or migration, feel free to ask or respond to this issue. I'm happy to help!

Kindest regards,

../Frenck

djansen1987, 2023-01-06:
HI @frenck,

Thanks for the message, i tried editing it but the energy dashboard now gives me the following error?

Entity has non-numeric state
The following entities have a state that cannot be parsed as a number:
sensor.esolar_sellelec (None)

if have setup the entity as following:
```
    SensorEntityDescription(
        key="totalElectricity",
        name="totalElectricity",
        icon="mdi:solar-panel-large",
        native_unit_of_measurement=ENERGY_KILO_WATT_HOUR,
        device_class=SensorDeviceClass.ENERGY,
        state_class=SensorStateClass.TOTAL_INCREASING,
    ),
```

```
            if self._type == 'totalElectricity':
                if 'totalElectricity' in energy['plantDetail']:
                    if energy['plantDetail']["totalElectricity"] is not None:
                        self._state = float(energy['plantDetail']["totalElectricity"])
```

if i log the value it is the following: 2023-01-06 10:51:36.748 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalElectricity State: 6283.61

Would you perhaps know how i can solve this ?

frenck, 2023-01-06:
> Thanks for the message, i tried editing it but the energy dashboard now gives me the following error?

That would not be caused by this change, as both `SensorDeviceClass.ENERGY` and `DEVICE_CLASS_ENERGY` do that same things (providing the string `energy`), as they are both a constant.

> Entity has non-numeric state
> The following entities have a state that cannot be parsed as a number:
> sensor.esolar_sellelec (None)

It probably means you end up sending `"None"` as a string as a state, instead of sending a `NoneType` non-string `None` value for the state.

../Frenck

djansen1987, 2023-01-06:
You are quick :) 

And you are right, found it couple of minutes after posting my reply. I would like to improve the addon even more. Move it from configuration.yaml to the UI but can't seem to some good examples, would you maybe have some tips on how to do this ?

frenck, 2023-01-06:
>  I would like to improve the addon even more.

Integration. Add-ons are different things 😉 

> Move it from configuration.yaml to the UI but can't seem to some good examples, would you maybe have some tips on how to do this ?

The best place to start on that, is just going to the Home Assistant Core codebase. Every integration that has an `config_flow.py` basically implemented this (and thus can be an example). This gives you instantly hundreds of examples :)

../Frenck

djansen1987, 2023-01-06:
Thanks will have a look!