djansen1987/SAJeSolar #51: Can you help me?

Herman2604, 2023-01-24:
Hi i am totaly new to all this and trying to install this SAJ monitor thing.
I have to copy all those files in a directory, but im not sure where .
Is this the right path? 
[image]

Ik hoop dat je kan helpen.
groetjes, Herman

djansen1987, 2023-01-25:
Hi Herman, 

Je heb 1x custom_components te veel. Nog makkelijker is het via HACS te installeren. This is een soort market waar je custom Components kan toevoegen die gemaakt zijn door gebruikers van HA. Let wel even op bij het installeren van mijn module dat er een PRE-release is. Deze zou wel moeten weten maar nog niet volledig doorgetest. 

Site van hacs
https://hacs.xyz/

Herman2604, 2023-01-27:
Ok bedankt!, ben toevallig net HACS aan het bekijken , het is allemaal nog erg nieuw.
Als ik je kan helpen met testen dan hoor ik het wel.

Grt Herman

https://hacs.xyz/

Herman2604, 2023-02-01:
OK , ik ben weer wat verder, heb nu alleen H1 sensor stukje in de Yaml file gezet, dat was zeker al de bedoeling geweest vanaf het begin? Ik heb nu een hele hoop sensors, moet alleen nog even controleren wat nu precies wat is.
Ik ga hier nog even mee spelen en dan is het tijd om dat biertje op te sturen :-)

djansen1987, 2023-02-10:
Hi, klinkt alsof je de goeie kant op gaat. Wat wel even belangrijk is, is dat je bij het installeren via HACS beta uit hebt staan. In beta staat een versie waarvan de handleiding afwijkt. 

In jou geval zou de yaml er ongeveer zo uit moeten zien:
```
sensor:
  - platform: saj_esolar
    username: aa@bb.cc
    password: abcd1234
    plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
    sensors: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid
      - chargeElec # Energy -> Home Battery Storage -> Energy going in to the battery (kWh)
      - dischargeElec # Energy -> Home Battery Storage -> Energy coming out of the battery (kWh)

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - isAlarm
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower
      - totalLoadPower
```

Herman2604, 2023-02-16:
Ja het begin is er,maar heb nog veel te leren, maar dat komt goed, bedankt voor de hulp 🙏
Kopje koffie is opgestuurd 😉

Grt Herman

Herman2604, 2023-02-20:
Het werkt trouwens alleen maar als ik deze regel weglaat:

plant_id: 0 # Default is 0.

Of mis ik iets?

Grt Herman

djansen1987, 2023-03-04:
> Ja het begin is er,maar heb nog veel te leren, maar dat komt goed, bedankt voor de hulp 🙏 Kopje koffie is opgestuurd 😉 Grt Herman

Hi Herman, Super tof. Thanks!

> Het werkt trouwens alleen maar als ik deze regel weglaat: plant_id: 0 # Default is 0. Of mis ik iets? Grt Herman

Ik heb dat eerder gehoord, Ik ga de broncode er even op naslaan. Ik ben wat heen en terug moeten gaan met code omdat er dingen niet meer werkte na een update, mogelijk dat deze gemist is in de laatste commit.