djansen1987/SAJeSolar #52: Error

Royalcyty, 2023-02-01:
Hola, soy un nuevo usuario de HA y Saj Solar, y quiero implementar tu trabajo pero me encuentro con varios errores y no se seguir con el trabajo
[image]

djansen1987, 2023-02-01:
Hola, no hablo español. 

Make sure you have not installed the Beta version of the intergration.

djansen1987, 2023-02-10:
Do you need any more help ?

Sendy16, 2023-02-27:
Ciao ho più o meno lo stesso errore
dopo aver installato tutto non riesco ad inserire nel file configuration.yaml le impostazioni....
mi da errore 

Error loading /config/configuration.yaml: while scanning a simple key
in "/config/configuration.yaml", line 65, column 5
could not find expected ':'
in "/config/configuration.yaml", line 66, column 5

Mi potreste aiutare ?

Sendy16, 2023-02-27:
qui il mio config

# Loads default set of integrations. Do not remove.
default_config:

# Load frontend themes from the themes folder
frontend:
  themes: !include_dir_merge_named themes

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

switch:
  - platform: template
    switches:
      ewelink_virtual_switch:
        turn_on:
          service: switch.turn_on
        turn_off:
          service: switch.turn_off

cover:
  - platform: template
    covers:
      ewelink_virtual_cover:
        open_cover:
          service: cover.open_cover
        close_cover:
          service: cover.close_cover
        stop_cover:
          service: cover.stop_cover
        set_cover_position:
          service: cover.set_cover_position

fan:
  - platform: template
    fans:
      ewelink_virtual_fan:
        value_template: "{{ states('input_boolean.state') }}"
        turn_on:
          service: fan.turn_on
        turn_off:
          service: fan.turn_off
        set_preset_mode:
          service: fan.set_preset_mode

light:
  - platform: template
    lights:
      ewelink_virtual_light:
        turn_on:
          service: light.turn_on
        turn_off:
          service: light.turn_off

sensor:
  - platform: saj_esolar
    username:-----
    password:-----
    plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
     sensors: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid
      - chargeElec # Energy -> Home Battery Storage -> Energy going in to the battery (kWh)
      - dischargeElec # Energy -> Home Battery Storage -> Energy coming out of the battery (kWh)

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - isAlarm
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower
      - totalLoadPower

djansen1987, 2023-03-04:
no response

closed 2023-03-04 by djansen1987