djansen1987/SAJeSolar #53: uitlezen SAJ converter

robbedoes14, 2023-02-11:
Hoi,

Ik heb een R5-12K-T2 SAJ omvormer met Aio3 module (met een Ethernet connectie in mijn router). Op de eSolar Portal kan ik inloggen en alle log gegevens bekijken.

Is het mogelijk om dit te integreren in Home Assistant ? Mij is niet helemaal duidelijk of dit nu wel of niet mogelijk is met deze integratie in HA.

Alvast dank.

djansen1987, 2023-02-11:
Hi Rob,

Het is mogelijk via de custom intergratie die ik hiervoor gemaakt heb. Let even op bij het installeren dat je niet kiest voor de beta versie wanneer je hem via Hacs installeerd. Wanneer je het handmatig doet, kies dan deze release: https://github.com/djansen1987/SAJeSolar/releases/tag/v1.3.1

Wanneer je geen batterij module of de SEC module in je meterkast heb komt je config er zo uit te zien:

```yaml
sensor:
  - platform: saj_esolar
    username: aa@bb.cc
    password: abcd1234
    plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - todayGridIncome
      - income
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - isAlarm # Yes / No
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower
      - systemPower # Installed capacity
```

djansen1987, 2023-02-12:
Hoi Rob,

Probeer volgende regel eens uit je config re halen

`plant_id: 0`

robbedoes14, 2023-02-12:
> 
aah, helaas......krijg nu de volgende foutmelding terug:

`[Invalid config for [sensor.saj_esolar]: value must be one of ['address', 'batCapcity', 'batCurr', 'batEnergyPercent', 'batteryDirection', 'batteryPower', 'buyElec', 'buyRate', 'chargeElec', 'currency', 'devOnlineNum', 'dischargeElec', 'gridDirection', 'gridPower', 'h1Online', 'income', 'isAlarm', 'isOnline', 'lastUploadTime', 'monthElectricity', 'nowPower', 'outPower', 'outPutDirection', 'peakPower', 'plantTreeNum', 'plantname', 'plantuid', 'pvDirection', 'pvElec', 'pvPower', 'reduceCo2', 'runningState', 'selfConsumedEnergy1', 'selfConsumedEnergy2', 'selfConsumedRate1', 'selfConsumedRate2', 'selfUseRate', 'sellElec', 'sellRate', 'solarPower', 'status', 'todayAlarmNum', 'todayElectricity', 'todayGridIncome', 'totalBuyElec', 'totalBuyEnergy', 'totalConsumpElec', 'totalElectricity', 'totalGridPower', 'totalLoadEnergy', 'totalLoadPower', 'totalPlantTreeNum', 'totalPvEnergy', 'totalPvgenPower', 'totalReduceCo2', 'totalSellElec', 'totalSellEnergy', 'useElec', 'yearElectricity'] @ data['resources'][19]. Got 'systemPower'. (See ?, line ?).](url)`

robbedoes14, 2023-02-16:
> Hoi Rob,
> 
> Probeer volgende regel eens uit je config re halen
> 
> `plant_id: 0`

Hoi Daniël, ik wil niet opdringerig zijn met mijn vraag omdat je ongetwijfeld veel tijd al hierin hebt gestopt en het natuurlijk niet op 'werk' moet gaan lijken. Maar ik weet ook niet of mijn vraag overkomt bij jou. Kan ik nog iets doen, of delen met jou zodat je mij wat meer richting kunt geven om voorstaande foutmelding te voorkomen. Zou het delen van mijn configuration.yaml met jou nog kunnen helpen of zijn er nog andere zaken die ik zou kunnen proberen. Zou ook graag jouw project willen sponsoren als dat bijdraagt, maar dan moet ik wel weten of mijn vraag overkomt....Alvast dank !

djansen1987, 2023-02-16:
Hoi Rob, had je reactie nog niet gezien sorry. En stel gerust je vragen. Privé momenteel wel wat druk met carnaval en baby op komst. Maar zal zien waar ik kan helpen.

Zou je de yaml config kunnen delen (zonder inloggen gegevens) daarnaast ook even goed om te weten welk release je hebt draaien. (In beta is alles anders en heb je alleen username en password nodig in de yaml config)

robbedoes14, 2023-02-16:
> Hoi Rob, had je reactie nog niet gezien sorry. En stel gerust je vragen. Privé momenteel wel wat druk met carnaval en baby op komst. Maar zal zien waar ik kan helpen.
>
> Zou je de yaml config kunnen [...]

Thnx, no worries, was even goed om te checken. Hieronder mijn configuration yaml:

`homeassistant:
  customize: !include customize.yaml

# Loads default set of integrations. Do not remove.
default_config:

# Load frontend themes from the themes folder
frontend:
  themes: !include_dir_merge_named themes

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

template:
  - sensor: 
      - name: hikvision_door
        state: "off"
      - name: hikvision_callstatus
        state: "off"
      - name: hikvision_motion
        state: "off"
      - name: hikvision_tamper
        state: "off"
      - name: hikvision_dismiss
        state: "off"

http:
# Path to your Cert Files
    ssl_certificate: /ssl/fullchain.pem
    ssl_key: /ssl/privkey.pem

# IP Ban
    ip_ban_enabled: true
    login_attempts_threshold: 5

# Example configuration.yaml entry
google_assistant:
    project_id: home-assistant-*****
    service_account: !include google_service_account.json
    report_state: true

# Afvalinfo    
sensor:
  - platform: afvalinfo
    resources:
      - type: gft
        friendly_name: Groene bak
      - type: pbd
        friendly_name: Oranje bak
      - type: papier
        friendly_name: Blauwe bak
      - type: restafval
        friendly_name: Grijze bak
      - type: trash_type_today
        friendly_name: Afval voor vandaag
      - type: trash_type_tomorrow
        friendly_name: Afval voor morgen
    location: ******
    postcode: ******
    streetnumber: 14
    dateformat: "%d-%m-%Y"
    locale: "nl"

  - platform: saj_esolar
    username: aa@bb.cc
    password: abcd1234
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - todayGridIncome
      - income
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - todayAlarmNum
      - isAlarm # Yes / No
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower
      - systemPower # Installed capacity`

Tav de release heb ik de volgende info gevonden:

[image]

Naar mijn idee heb ik niet de beta geinstalleerd. Zijn er nog meer manieren om de release te achterhalen die ik nu draai ? Overigens staat er bij de username in het voorbeeld nu een @ opgenomen. In mijn geval heb ik daar geen mailadres staan maar mijn login naam die ik ook gebruik om in te loggen op de web-portal van eSolar.

gr. Rob

robbedoes14, 2023-02-16:
wellicht beter zichtbaar met een plaatje:

[image]

djansen1987, 2023-02-16:
Bedankt, zou het kunnen zijn dat je een reseller account hebt met meerdere plants ?

Wat je zou kunnen doen is debug aanzetten van deze module via de yaml, in je HA debug log komt dan de json data die de eSolar api terug geeft (of foutmelding). Dit log zou je eventueel kunnen delen met me via GitHub@djansen.nl . Reden van mailen is dat er mogelijk ook wat privé gegevens of plant details in staan.

djansen1987, 2023-02-16:
Je release lijkt overigens wel goed

robbedoes14, 2023-02-17:
> Bedankt, zou het kunnen zijn dat je een reseller account hebt met meerdere plants ?
> 
> Wat je zou kunnen doen is debug aanzetten van deze module via de yaml, in je HA debug log komt dan de json data die de eSolar api terug geeft (of foutmelding). Dit log zou je eventueel kunnen delen met me via [GitHub@djansen.nl](mailto:GitHub@djansen.nl) . Reden van mailen is dat er mogelijk ook wat privé gegevens of plant details in staan.

ik heb je gemaild. Fijne carnaval !

Beaker73, 2023-02-24:
Ikzelf heb deze vandaag ook geinstalleerd en liep tegen dezelfde problemen aan.

Voor het laatste probleem moest ik de laatste regel met
- systemPower verwijderen.

Daarna werkte het bij mij zonder problemen.

djansen1987, 2023-03-04:
Via de mail begreep ik dat dit voor rob ook de oplossing was. Super bedankt ook voor je sponsering!!! Ik zal de handleiding even naloop en kijken of ik de uitleg weer strak krijg. Veel plezier bij het gebruik van de intergratie!

closed 2023-03-04 by djansen1987