djansen1987/SAJeSolar #60: Add more states to runningState handling

elboletaire, 2023-04-29:
For what I can see in the code, the state handling is binary: if it's zero it's shown as `off`, any other value is shown as `on`. In my case, I see I'm getting the state as `2` and, in the official SAJ portal, I can see the state as red. It doesn't mean it's not working, but there's some error connection in the inverter (a red color isn't a good UX for this case).

[image]
[image]

I'm pretty newbie to homeassistant and, more specifically, to how plugins and sensors are done but, could it be possible to make it non binary? In this case, I'd show a "warning" text for this specific case.

djansen1987, 2023-08-29:
Hi @elboletaire , what kind of device do you have? as there are multiple status values. also for example for accessories to converts

elboletaire, 2023-08-30:
TBH I ended up modifying entirely the plugin on my own, but the change looks good, so I'll close the issue.

closed 2023-08-30 by elboletaire

djansen1987, 2023-08-30:
That is sad to hear, you could also do a PR to make the integration better for everyone. But thank you for the confirmation.

elboletaire, 2023-08-30:
> That is sad to hear, you could also do a PR to make the integration better for everyone. But thank you for the confirmation.

Not sure my edits were worth adding... after doing many changes I realized I could had avoided doing changes by just creating aliases... If I were to do changes to the plugin I think I'd change it entirely, grouping sensors in a better way. But that's a lot of work and, as it is right now works for me, so.. 🤷🏼

djansen1987, 2023-08-30:
I understand 👍 if you ever have some great additions let me know. I self do not use the integration my self an created it for a family member.  For him it is also working so any additions had low priority unleas added via PR