djansen1987/SAJeSolar #64: Add uuid param to reduce calls load and allow unpaired users to use the plugin

elboletaire, 2023-10-07:
Today I was migrating home assistant from a docker to a raspberry pi. Something I tried twice before, and had no issues, but today suddenly, after loading the backup in the raspberry pi, I noticed all the esolar sensors weren't working.

After debugging some time, I realized my plant was unlinked from my account in the esolar website, but I can still access it via the URL

    https://fop.saj-electric.com/saj/monitor/site/detailForm?plantuid={UUID}

I'm aware this plugin can't be the culprit, unless saj website has some serious issue in the used calls, and they unlinked my account when migrating (very unlikely, and remember I tried migrating twice this week without issues).

So I manually hardcoded the UUID in the `sensor.py` file in order to have it working. I do realize I lost some sensors that were taken from the plant list, but I'm not really using those, so I don't care.

I think you could add a configurable way of adding that UUID, and avoid the `response2` entirely in such case (or find other means to get those values, without the plant list requirement). Obviously this should be accompanied with a notice warning about the possible missing sensors, in case you opt for that way.

djansen1987, 2023-10-08:
are you perhaps a greenheiss user ? If so please read here: https://github.com/djansen1987/SAJeSolar/issues/65

elboletaire, 2023-10-09:
No, but it's the same issue. I'm from Solarprofit and the last comment adds a link to their manager (inversor.saj-electric.com). And login there I can see the plant. So it's probably a better idea to just add means to configure the domain, rather than adding the uuid feature I was requesting.

djansen1987, 2023-10-17:
fixed in https://github.com/djansen1987/SAJeSolar/releases/tag/v1.5.1

closed 2023-10-17 by djansen1987