djansen1987/SAJeSolar #65: is fop.saj-electric.com deprecated?

santiagozky, 2023-10-07:
I noticed that the integration was not working anymore for me. I accessed the esolar site and noticed that
-My plant was gone. no more data
-There is a button "new platform" taking me to https://esaj-home.saj-electric.com/index. That new site does shows my data.

Did anyone else experienced this?

A bit more info...

My R5 inversor and esolar module are rebranded with the "Greenheiss" brand (for Spain market). Their version of esolar (which is basically a reskin of SAJ's) still shows my plant https://inversores-style.greenheiss.com/cloud. if anyone's elses integration got killed it might be worth to check that domain.

My guess is they are deprecating the fop site but left Greenheiss users in the old one?

in any case, the integration is probably not working anymore for greenheiss  users (and probably others rebranded units ) unless we point to their domain (maybe it can be configured? I'll try to check).

santiagozky, 2023-10-07:
I managed to make it work temporarily by:
- replacing the  domain in the sensor.py
- forcing it not to verify SSL certificates (seems like the cert chain is broken and HASS rejects it):
session = async_create_clientsession(hass,False).

teamMOYA, 2023-10-07:
The same is happening to me. My plant is gone in the "old platform".

miguelzx, 2023-10-07:
> I managed to make it work temporarily by:
> 
>     * replacing the  domain in the sensor.py
> 
>     * forcing it not to verify SSL certificates (seems like the cert chain is broken and HASS rejects it):
>       session = async_create_clientsession(hass,False).

Hola puede explicarlo mejor, es que no lo pillo gracias

santiagozky, 2023-10-08:
@teamMOYA is your kit branded as SAJ or some other white label (eg. Greenheiss)?

arnauplan, 2023-10-08:
Hi, the exact same thing is happening to me but I did what you said and it still doesn't work. Can you explain exactly what needs to be changed?
I have GreenHeiss.
Thanks!

teamMOYA, 2023-10-08:
> @teamMOYA is your kit branded as SAJ or some other white label (eg. Greenheiss)?

Mine are rebranded as "Solarprofit", but they don't have a web like Saj and Greenheiss. I'll have to wait for the new web access update.

arnauplan, 2023-10-08:
> I managed to make it work temporarily by:
> 
> * replacing the  domain in the sensor.py
> * forcing it not to verify SSL certificates (seems like the cert chain is broken and HASS rejects it):
>   session = async_create_clientsession(hass,False).

Where do you change the domain to inside sensor.py?
Do you change https://fop.saj-electric.com/saj to https://esaj-home.saj-electric.com/ or something?

miguelzx, 2023-10-08:
Hello, I also had that problem.

My Greenheiss h1 inverter. I use an old integration 1.3.1 which I already have working.

Right now everything works perfectly for me

replacing the domain in the sensor.py

I leave a copy of my sensor.py, in case it helps you, but remember that I use 1.3.1

"""
Alternative for the SAJ local API sensor. Unfortunally there is no public api.
This Sensor will read the private api of the eSolar portal at https://fop.saj-electric.com/
"""

import asyncio
from homeassistant.helpers.aiohttp_client import async_create_clientsession
from datetime import timedelta
import datetime
import calendar

from functools import reduce
import logging
from typing import Final

import aiohttp
import async_timeout
import voluptuous as vol

import re
from bs4 import BeautifulSoup

from homeassistant.components.sensor import (
    PLATFORM_SCHEMA,
    STATE_CLASS_TOTAL_INCREASING,
    SensorEntity,
    SensorEntityDescription,
)
from homeassistant.const import (
    CONF_RESOURCES,
    CONF_USERNAME,
    CONF_PASSWORD,
    CONF_SENSORS,
    CONF_DEVICE_ID,
    DEVICE_CLASS_ENERGY,
    DEVICE_CLASS_POWER,
    ENERGY_KILO_WATT_HOUR,
    POWER_WATT,
    PERCENTAGE,

)
from homeassistant.helpers.aiohttp_client import async_get_clientsession
import homeassistant.helpers.config_validation as cv
from homeassistant.helpers.entity import Entity
from homeassistant.util import Throttle, dt

def add_months(sourcedate, months):
    month = sourcedate.month - 1 + months
    year = sourcedate.year + month // 12
    month = month % 12 + 1
    day = min(sourcedate.day, calendar.monthrange(year,month)[1])
    return datetime.date(year, month, day)

def add_years(d, years):
    try:
        return d.replace(year = d.year + years)
    except ValueError:
        return d + (date(d.year + years, 1, 1) - date(d.year, 1, 1))

BASE_URL = 'https://inversores-style.greenheiss.com/cloud/login'
[... 1423 lines cut ...]
                if self._type == 'totalPvgenPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][4][-1])

                # getPlantMeterDetailInfo
                if self._type == 'totalPvEnergy':
                    if 'totalPvEnergy' in energy["getPlantMeterDetailInfo"]['plantDetail']:
                        if energy["getPlantMeterDetailInfo"]['plantDetail']["totalPvEnergy"] is not None:
                            self._state = (energy["getPlantMeterDetailInfo"]['plantDetail']["totalPvEnergy"])
                if self._type == 'totalLoadEnergy':
                    if 'totalLoadEnergy' in energy["getPlantMeterDetailInfo"]['plantDetail']:
                        if energy["getPlantMeterDetailInfo"]['plantDetail']["totalLoadEnergy"] is not None:
                            self._state = (energy["getPlantMeterDetailInfo"]['plantDetail']["totalLoadEnergy"])
                if self._type == 'totalBuyEnergy':
                    if 'totalBuyEnergy' in energy["getPlantMeterDetailInfo"]['plantDetail']:
                        if energy["getPlantMeterDetailInfo"]['plantDetail']["totalBuyEnergy"] is not None:
                            self._state = (energy["getPlantMeterDetailInfo"]['plantDetail']["totalBuyEnergy"])
                if self._type == 'totalSellEnergy':
                    if 'totalSellEnergy' in energy["getPlantMeterDetailInfo"]['plantDetail']:
                        if energy["getPlantMeterDetailInfo"]['plantDetail']["totalSellEnergy"] is not None:
                            self._state = (energy["getPlantMeterDetailInfo"]['plantDetail']["totalSellEnergy"])

            # -Debug- adding sensor
            _LOGGER.debug(f"Device: {self._type} State: {self._state}")

santiagozky, 2023-10-08:
@teamMOYA check if by any chance your credentials work on the greenheiss url mentioned in my first comment. you might be lucky and could use the modified version that @miguelzx posted (which I guess is the same I did)

santiagozky, 2023-10-08:
@teamMOYA seems like solar the profit login is at https://inversor.saj-electric.com/cloud?login

my credentials work as well there. pretty sure they just reskin the system with different logos and colors. you probably can use the greenheiss url or this one.

miguelzx, 2023-10-08:
[](url)

> @teamMOYA check if by any chance your credentials work on the greenheiss url mentioned in my first comment. you might be lucky and could use the modified version that @miguelzx posted (which I guess is the same I did)

Hello, yes that's correct, that's what you said, thanks for the help.

That is the file as I have it, in case it is of any help to you.

Since yesterday I spent a hellish afternoon. Restoring backups, and thinking that the problem is that I had updated home assistant

djansen1987, 2023-10-08:
i have a couple of account to test and develop with but all of them seem to work on the original portal so i am guessing it is only the greenheiss that stopped working. If there is anyone willing to share their login details i will see if i can add a option to the configuration yaml to switch to the correct url.

arnauplan, 2023-10-08:
> Hello, I also had that problem.
>
> My Greenheiss h1 inverter. I use an old integration 1.3.1 which I already have working.
>
> Right now everything works perfectly for me
>
> replacing the domain in the [...]

Thank you!!

djansen1987, 2023-10-08:
@arnauplan @santiagozky @teamMOYA 

**Please be aware** that the sensor.py above has more changes then only the domainname/url. **!!Use at your own risk!!**.

@miguelzx could you explain why you have change the H1 part and use beautiful soup to do this? Per my understanding the use of beautiful soup is not permitted by home assistant.

djansen1987, 2023-10-08:
> The same is happening to me. My plant is gone in the "old platform".

Is your plant available in the new portal or do you have to go to the greenheiss portal ?

teamMOYA, 2023-10-08:
> Is your plant available in the new portal or do you have to go to the greenheiss portal ?

 

- https://esaj-home.saj-electric.com/ - available
- https://inversor.saj-electric.com/ - available in "shared plants" but only general data. I can't see the detailed data of the inversor+battery
- https://fop.saj-electric.com/saj - no plant/no data
- https://inversores-style.greenheiss.com/ - no plant/no data

elboletaire, 2023-10-09:
> > @teamMOYA is your kit branded as SAJ or some other white label (eg. Greenheiss)?
> 
> Mine are rebranded as "Solarprofit", but they don't have a web like Saj and Greenheiss. I'll have to wait for the new web access update.

You can still use any of the saj websites with your Solarprofit credentials.

So I guess having a way to configure our endpoint in the addon should suffice, although some websites may have different structure or calls, and that could complicate things.

santiagoGPNC, 2023-10-09:
@teamMOYA  maybe you can login to one of the other ones and re add your inverter and esolar module there?

santiagozky, 2023-10-10:
@djansen1987  I could provide you my credentials if that would help

Txetxo1979, 2023-10-10:
Has the easiest way been found to be able to see the sensors again? I am new to home assistant.

santiagozky, 2023-10-10:
@Txetxo1979  so far, it seems to find out in which portal you can find your info (greenheiss, solarprofit, etc) and then modify the sensor.py file of the extension to replace the SAJ url with the one that works for you.

djansen1987, 2023-10-10:
> @djansen1987  I could provide you my credentials if that would help

That would help a lot. Could you send it to github@djansen.nl promise to handle it with care!

Txetxo1979, 2023-10-10:
> @Txetxo1979 so far, it seems to find out in which portal you can find your info (greenheiss, solarprofit, etc) and then modify the sensor.py file of the extension to replace the SAJ url with the one that works for you.

My URL is https://esaj-home.saj-electric.com/login
but I don't know where I can go to change the extension in sensor.py

djansen1987, 2023-10-10:
> My URL is https://esaj-home.saj-electric.com/login but I don't know where I can go to change the extension in sensor.py

This is the new portal, the api is completely changed on that site. Implementing that is almost like starting from scratch.. don't know if i got time for this soon.

miguelzx, 2023-10-10:
> @arnauplan @santiagozky @teamMOYA
> 
> **Please be aware** that the sensor.py above has more changes then only the domainname/url. **!!Use at your own risk!!**.
> 
> @miguelzx could you explain why you have change the H1 part and use beautiful soup to do this? Per my understanding the use of beautiful soup is not permitted by home assistant.
> 
> addition: the code used is a very old version of my custom integration. It still uses the old energy StateClass which are not supported by home assistant

Hello, my language is not English, so I hope the translator works well.

The version that is theirs, which as I mentioned is 1.3.1, which has worked well for me, until that happened. I use the latest version of home assistan and with my inverter, which is a saj clone, it works perfectly.

What I did was put it in sensor.py, it was replace

https://inversor.saj-electric.com/cloud?login, by https://inversores-style.greenheiss.com/cloud/logou.

and this session = async_create_clientsession(hass,False)

What did the previous colleague say?

Txetxo1979, 2023-10-11:
> This is the new portal, the api is completely changed on that site. Implementing that is almost like starting from scratch.. don't know if i got time for this soon.

Thank you so much.
I just saw that I also have the information at https://inversor.saj-electric.com/cloud?login
What should I do then to be able to have the information in home assistant?

santiagozky, 2023-10-11:
@miguelzx  
Hola Miguel,
Yo tambien uso 1.3.1. no he actualizado a la nueva version.
el codigo tiene puesto directamente el dominio de saj en varios lugares. Tengo una version que lo actualiza con el dominio de greenheiss.

CUIDADO:tiene algunos cambios mas porque en mi caso que no tengo bateria, reportaba algunos valores en el sensor equivocado (despues de la linea 1136).
Anexo mi fichero tal cual lo tengo. Compararlo con el fichero que tienes en ` custom_components/saj_esolar/sensor.py` de tu instalación de Home Assistant.
mi sensor.py tal cual lo tengo. it https://gist.github.com/santiagozky/311e830cffd0b54d2fa115162c5ad73c

santiagozky, 2023-10-11:
@Txetxo1979  if you dont mind reverting to 1.3.1. (which is what I still use), you could adjust my gist above to point to your url. just adjust these lines:

BASE_HOST = 'inversores-style.greenheiss.com'
BASE_DOMAIN = 'https://'+BASE_HOST
BASE_URL = BASE_DOMAIN+'/cloud'

caution: I did additional changes between lines 1138 and 1150 because of a mismatch in my data, probably because I dont have a battery  (see https://github.com/djansen1987/SAJeSolar/issues/41), you could skip those changes.

elboletaire, 2023-10-11:
@santiagozky why are you recommending people downgrading and updating a lower version? Latest version can be changed and fixed the same way 😕

santiagozky, 2023-10-11:
@elboletaire  I havent recommended anyone to downgrade. Im merely offering the solution I have for myself. Im on 1.3.1 because I had previous changes that I dont want to reapply on newer versions.

78lobo78, 2023-10-14:
Hello, when will the esolar platform be updated so that the sensors work well?

santiagozky, 2023-10-16:
I tried making a change but I've never done anything in python or HASS, so it's taking me quite a long.
I might keep giving it a shot, but I cannot make promises on how long it will take me. for those waitings, if you have a minimum programming experience you might be able to adjust it manually replacing all the urls in the sensor.py file.

matly006, 2023-10-16:
Good evening. I have an h1 inverter and valid credentials for both the https://esaj-home.saj-electric.com/index portal and for the old https://fop.saj-electric.com/saj.

If I can help, I'd be happy to help

djansen1987, 2023-10-17:
Hi All, thanks to the work of @santiagozky this should now working in the latest version. Please read the release notes to known what to do. Thanks all for the input and patients! 

https://github.com/djansen1987/SAJeSolar/releases/tag/v1.5.1

santiagozky, 2023-10-17:
thank you for merging and releasing @djansen1987

closed 2023-10-17 by santiagozky