djansen1987/SAJeSolar #72: All data lose -New platform migrate.

TecnoTalarn, 2023-12-08:
Hello,
I have been using this software for over a year without any problems.
Today, December 8th, I lost all the information I was collecting without any prior warning.
I thought my HA was corrupted, and pulled the backup. Same result, I don't see the battery, I don't see any information from the SAJ portal.
Looking at the manufacturer's app, everything is OK. Finally, I connect to the SAJ portal and I don't see my investor or any data. I see that there is a switch to a "new platform", I enter that platform and there is all the information.
I already understood that it happened. They have migrated my plant to the new portal, without warning.
I imagine it's happening to all users. And those of us who depend on HA are now blind. In my case, I need to know the battery charge to decide if I should connect the inverter to the grid, since I am usually not connected to it.
Do you know if there is a way to fix this quickly? If it had happened in the middle of summer, I wouldn't worry too much, but at this time, I need to be aware of how the battery is doing.

Thanks.

TecnoTalarn, 2023-12-08:
Well, I've tried without success since I'm not very good at application programming, to modify the URL where it connects but it's not enough. It would be possible to modify the way of passing the login data, and also, for sure, it would not be able to capture the data that is on the web.
The maximum I have achieved is to put the URL of the login.
2023-12-09 00:36:56.255 ERROR (MainThread) [custom_components.saj_esolar.sensor] https://esaj-home.saj-electric.com/login returned 405

stefagi, 2023-12-08:
I have the same problem.
I'm trying to use the url "inversor.saj-electric.com" as described here: https://github.com/djansen1987/SAJeSolar/releases/tag/v1.5.1
without success

TecnoTalarn, 2023-12-09:
Well, I have seen that the greenheiss or solarprofit inverters are still operational on the old platform, but the SAJ devices are no longer there. I tried to connect to the greenheiss platform with my user and it can be accessed, but there is no inverter and if you try to add an inverter SAJ detects a wrong SN.
What I haven't tried is re-enabling the inverter on the old SAJ website, because I think they would delete it immediately.

I hope someone more knowledgeable than me can find a way to access the new platform soon!

TecnoTalarn, 2023-12-09:
https://esaj-home.saj-electric.com/monitor/device-monitor-detail?deviceSn=XXXXXXXXXXXX
This is the only page that is needed. Maybe the API can go much better, but if we could get the info from here, it would be fine.

This other page also works, at the moment:
https://inversor.saj-electric.com/cloud/cloudMonitor/deviceInfo?devicesn=XXXXXXXXXXXX
But I don't see how to edit the component so that it goes directly to read this.

palmfieldsflame, 2023-12-09:
I had the same problem from yesterday but mine has started working again just now with no changes to the the configuration. Hope it stays working.

miquelbotanch, 2023-12-09:
Here in Barcelona, it has been giving me strange values during a day or so, after that, all of a sudden the integration returned me "no numeric value" for grid production and battery level, and now, it's working again.

Maybe I'm getting the info from another server or SAJ changed something
[image]

TecnoTalarn, 2023-12-10:
Hello again
I made changes on my module to connect a SOLARPROFIT page, and run fine.
Also I see all data are again in esolar fop webserver. Maybe a database error on server, and have resolved.

If anyone need solarprofit mod, I can send. My version is very old, but I made a lot of changes in order to see more data from server, like independent array values or negative power value on battery to see charge or discharge.

Thanks from Catalonia.

palmfieldsflame, 2023-12-22:
My HA was not pulling data from the SAJ portal again today. I looked here and there seemed to be no one else facing the same issue as me. I then checked my HA logs and found these errors.
[image]
[image]

After some investigation It turns out that my data is on the international node and the urls for the international and European nodes are different also the url for the European node is the same as the old url.
The url for the international node is https://intop.saj-electric.com/saj/login. Searched for 'https://fop.saj-electric.com/saj/login' and replaced it with 'https://intop.saj-electric.com/saj/login' in sensor.py and now my data is back.

OlorinSledge, 2023-12-25:
> My HA was not pulling data from the SAJ portal again today. I looked here and there seemed to be no one else facing the same issue as me. I then checked my HA logs and found these errors. ![Error1](https://private-user-images.githubusercontent.com/93476042/292348716-749db014-440d-4416-bbe6-04c36a701795.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM1NDE5NDYsIm5iZiI6MTcwMzU0MTY0NiwicGF0aCI6Ii85MzQ3NjA0Mi8yOTIzNDg3MTYtNzQ5ZGIwMTQtNDQwZC00NDE2LWJiZTYtMDRjMzZhNzAxNzk1LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI1VDIyMDA0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkMWYzOWViMWVlZWYwODM3ZjJhZDM2MmIxNjNkMjY2MjgxMjViOWFlNzhjYTE2Nzc5MGY3MGRjY2I4MWVmMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.P5J28IfNXr4Ka6hS9cX6fyztYU-1_t0ZGnFBXnTuJKE) ![error2](https://private-user-images.githubusercontent.com/93476042/292348739-b094e845-160c-451f-9f8f-dc6dad9990a6.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM1NDE5NDYsIm5iZiI6MTcwMzU0MTY0NiwicGF0aCI6Ii85MzQ3NjA0Mi8yOTIzNDg3MzktYjA5NGU4NDUtMTYwYy00NTFmLTlmOGYtZGM2ZGFkOTk5MGE2LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI1VDIyMDA0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE0OTE2ZmI1NTlkMjdmYzE2ZTJhNDdlMzE0ODU4NmU2OTc2ZTViMTQ3MjUyNDg0ZDg1NzJkZjkzMTRjNDRiYzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Y-iITW6guEIRgJu2DKlTLSr7HqmhMlE4bCW_oGogNl0)
> 
> After some investigation It turns out that my data is on the international node and the urls for the international and European nodes are different also the url for the European node is the same as the old url. The url for the international node is https://intop.saj-electric.com/saj/login. Searched for 'https://fop.saj-electric.com/saj/login' and replaced it with 'https://intop.saj-electric.com/saj/login' in sensor.py and now my data is back.

I too am now using the international node. You need to search for all instances of "fop" and replace with "intop".  There are two locations in sensor.py:

1. Line 55: BASE_URL = 'https://fop.saj-electric.com/saj/login'
2. Line 506: vol.Optional("provider_domain",default="fop.saj-electric.com"): cv.string,

Change the "fop" to "intop" for both the above lines and everything will be working again.

djansen1987, 2024-01-05:
If anyone can make a pull request so it works in both situations i am happy to merge it. Unfortuanly i also do not have a saj device anymore and there for it is not on top of my priorities. Also happy to assign a second owner to the projects

santiagoGPNC, 2024-01-16:
@OlorinSledge  @palmfieldsflame  in the most recent version of the code, the domain can be passed through the configuration . This was done for the benefit of solarprofit and greenheiss users, but I think should work for this as well.

e.g. adding

```
provider_domain: https://intop.saj-electric.com
provider_path: saj
```

will most likely do the same as you mention

santiagoGPNC, 2024-01-19:
@connornz  you should be able to change the domain in your configuration.yaml following the configuration instructions

>provider_domain: https://intop.saj-electric.com
provider_path: saj

connornz, 2024-01-19:
> @connornz you should be able to change the domain in your configuration.yaml following the configuration instructions
> 
> > provider_domain: https://intop.saj-electric.com
> > provider_path: saj

Like this? 

#sensor:
  - platform: saj_esolar
    username: connor_nz
    password: PASSWORD
    plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
    provider_domain: https://intop.saj-electric.com
    provider_path: saj
    sensor: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - todayGridIncome
      - income
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower
      - systemPower # Installed capacity

santiagoGPNC, 2024-01-19:
on a glance, yes (assuming you have the most recent version, that is.)

connornz, 2024-01-19:
> on a glance, yes (assuming you have the most recent version, that is.)

Doesn't matter what I do. I get the following 
"ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE"

djansen1987, 2024-04-07:
Did you get it resolved ?

connornz, 2024-04-07:
Nope. But haven’t tried recently. I manually specified the Int domain as I use the international server but still no luck. Given up for now. Using modbus to read battery percentage, but usage No enterties for battery that I can use in the energy dashboard.

djansen1987, 2024-04-08:
I also want to look in to the Modbus. Could you perhaps share how you connected the modbus and what device you use to read the values in Home assistant ?

djansen1987, 2024-08-09:
No response

closed 2024-08-09 by djansen1987