djansen1987/SAJeSolar #74: SAJ Portal Server changes

bowenda, 2024-01-29:
In Late December 2023, SAJ changed their portal server, there are now several regional severs.

How do I change the server used from https://fop.saj-electric.com/ to https://intop.saj-electric.com/saj. This is the international Server.

Additionally the Plant ID is now a 6 character ASCII string which fails the configuration check.

Thanks
David

djansen1987, 2024-02-19:
please use this part from the documentation to update the url 

```yaml
    provider_domain: inverters.resellerdomain.ext
    provider_path: cloud
```

https://github.com/djansen1987/SAJeSolar

djansen1987, 2024-02-19:
for the plant id you can try to edit your sensor.py in the custom_components folder from int to string

`        vol.Optional(CONF_PLANT_ID, default=0): cv.positive_int, # type: ignore`
https://github.com/djansen1987/SAJeSolar/blob/90cf5e55372e621f99506ede2477e0185cc2f17d/custom_components/saj_esolar/sensor.py#L502

This will allow you to enter an ASCII string instead of only a number. Let me know if that solves it so i can see if we need to update the source code on github

manducagiuseppe, 2024-02-29:
if it can be here are the profits in the new servers
[image]

closed 2024-04-07 by djansen1987