djansen1987/SAJeSolar #75: warnings on deprecated units

Pierric82, 2024-02-14:
Hi,

Just wanted to report HA told me this today when restarting the container:
```
2024-02-14T12:08:28.797946052Z 2024-02-14 13:08:28.795 WARNING (MainThread) [homeassistant.const] ENERGY_KILO_WATT_HOUR was used from saj_esolar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/djansen1987/SAJeSolar/issues

2024-02-14T12:08:28.819433336Z 2024-02-14 13:08:28.810 WARNING (MainThread) [homeassistant.const] POWER_WATT was used from saj_esolar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/djansen1987/SAJeSolar/issues
```

So I thought I'd be a good citizen and open this issue (haven't seen any other references to this)

djansen1987, 2024-02-19:
thanks, will update this in the following release. You may also do a pull requests if you want to get it resolved sooner

LAdegeest, 2024-04-05:
Hi,
I've solved the issue locally, but I've no experience with pull request. Do I have to create a branch first?

djansen1987, 2024-04-07:
Hi, Great that you want to help!

first you have to "Fork" this project on github. Like below
[image]

After this a copy of the code is made in to your own github. In this copy you can make changes to the code (github desktop with vscode are good software options, or direct in the browser). After you saved (commit) the code  go to your own github repository and click new pull request:
[image]
This will show the changes you made. Check if correct and click Create Pull Request. In my example i added the word Test to a readme.md.
[image]
After this i get a notification, i will then also review the code and if okay i will merge it in the existing code.

Hope this helps and will look our for your pull request, awesome that you want to help the project! Thanks

djansen1987, 2024-06-24:
This is fixed by @LAdegeest in the latest release. Thanks for reporting!

closed 2024-06-24 by djansen1987