djansen1987/SAJeSolar #79: Help new system

1achy, 2024-03-09:
I need help i install solar system saj h1-6k-s3 inverte and use 👍 

# Configure SENSORE scadenza certificato integraazione

# impianto solare su cluod      
  - platform: saj_esolar
    username: xxxx
    password: xxxx
    provider_domain: https://fop.saj-electric.com/
    provider_path: saj
    plant_id: 61S8PC # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
    sensors: h1
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity # Energy -> Solar production
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - devOnlineNum
      - selfUseRate
      - totalBuyElec # Energy -> Grid consumption
      - totalConsumpElec
      - totalSellElec # Energy -> Return to grid
      - chargeElec # Energy -> Home Battery Storage -> Energy going in to the battery (kWh)
      - dischargeElec # Energy -> Home Battery Storage -> Energy coming out of the battery (kWh)

      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2
      - batCapcity
      - batCurr
      - batEnergyPercent
      - batteryDirection
      - batteryPower
      - gridDirection
      - gridPower
      - h1Online
      - outPower
      - outPutDirection
      - pvDirection
      - pvPower
      - solarPower
      - totalLoadPower

i cant see nothing

1achy, 2024-03-09:
log:

2024-03-09 13:07:36.498 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.499 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.499 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.499 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.500 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.500 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.500 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2024-03-09 13:07:36.500 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE

1achy, 2024-03-09:
i solved but i think the value is no correct : 

[image]

djansen1987, 2024-04-07:
It could take a few hour to settle. Is it now working as it should ?

Th4nosIt, 2024-07-13:
Hi...
Does anyone know how to get Battery Temperature? Not listed above.
Thanks

djansen1987, 2024-08-09:
Maybe this is in the coming release, else maybe this can be contributed. If not in new release please create separate request for it

closed 2024-08-09 by djansen1987