djansen1987/SAJeSolar #81: Incorrect Reporting of totalGridPower and totalLoadPower in SAJ eSolar Integration

LowKey88, 2024-03-23:
While debugging the SAJ eSolar custom component, I noticed that the totalGridPower and totalLoadPower metrics are consistently reported as 0.0, which seems incorrect. This issue occurs in the context of fetching data for my solar energy system.

Upon closer inspection through the web interface (via Inspect Element), the actual values for these metrics are shown correctly (e.g., totalGridPower: 2507.0, totalLoadPower: 2507.0 on 2024-03-24 01:40:00). It appears that the integration may not be parsing or handling the data correctly from the API response.

Debug logs show the following for these metrics:

```
2024-03-24 01:46:05.657 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalGridPower State: 0.0
2024-03-24 01:46:05.657 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalLoadPower State: 0.0
```

However, the API response clearly provides non-zero values:

```
"totalGridPower": 2507.0,
"totalLoadPower": 2507.0,
...
```

After upgrading my inverter from the R5 series to the R6 series (R6-20K-T2-32), I continued using the same SAJ Sec Module (M5370G2025XXXXXX). It's possible that this hardware change might be influencing how data is reported or interpreted by the integration, leading to the observed discrepancies in totalGridPower and totalLoadPower metrics.

Including this information will help developers to better understand the potential impact of hardware changes on the integration's functionality.

djansen1987, 2024-04-07:
In the debug log in HA also the complete json is printed. Do you see the values in the json or are they also 0.0 there.
If so, it could be that they have changed their api. Could you also provide the API url where you have found the corrent information?

djansen1987, 2024-08-09:
No response

closed 2024-08-09 by djansen1987

santiagozky, 2024-10-07:
I think this is what I originally asked around here:
https://github.com/djansen1987/SAJeSolar/issues/41

I think there is a confusion on what "totalloadpower" is meant to be.
currently, at least for saj_sec, totalloadpower is taken from column 2. 
looking at the data and website I think column is the  "self consumed" energy (i.e. amount of the solar power you are consuming) (so, at night it will always be zeo).
column 1 has the total consumption of the come (no matter if it comes from grid or panels).
This is consistent with the UI as well:
[image]

since my first bug didn't reach to any conclusion (at the time it seems I was the only sec user ) I just changed the line by myself, but maybe now it's more clear this is incorrect.

what I did was simply changing the line for totalLoadPower to pick column 1 instead of 2.
```
 if self._type == 'totalLoadPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][2][-1])
```

to 

```
 if self._type == 'totalLoadPower':
                    if 'dataCountList' in energy:
                        if energy["getPlantMeterChartData"]['dataCountList'][4][-1] is not None:
                            self._state = float(energy["getPlantMeterChartData"]['dataCountList'][1][-1])
```

santiagozky, 2024-10-08:
@LowKey88  since this bug is closed, maybe you can checkout my PR at https://github.com/djansen1987/SAJeSolar/pull/96#issuecomment-2396750969 and see if you think this explains the behaviour? maybe you can try my changes? I only fixed totalLoadPower, but I think I figured out the issue, but I would love some confirmation.