djansen1987/SAJeSolar #84: Error SAJ_SOLAR  HAC

AGC1964, 2024-06-26:
Tryin to include SAJ Solar panel in HASS and getting errors.  

Registrador: homeassistant.setup
Fuente: setup.py:349
Ocurrió por primera vez: 12:12:11 (1 ocurrencias)
Último inicio de sesión: 12:12:11

The 'saj_esolar' integration does not support YAML setup, please remove it from your configuration

Registrador: homeassistant.setup
Fuente: setup.py:269
Ocurrió por primera vez: 12:12:11 (1 ocurrencias)
Último inicio de sesión: 12:12:11

Setup failed for custom integration 'saj_esolar': No setup or config entry setup function defined.

I have a GH-I 8.0 2M STYLE

Thanks for the help

A

djansen1987, 2024-06-26:
I think you first have to reboot after installing the hacs. After the reboot of hacs then add the yaml. If the error still occurs please add the yaml here so I can check what is wrong with it. Please be sure NOT include username password. So REMOVE from yaml before posting 😉

AGC1964, 2024-06-26:
Thanks for the fast reply.  

I deleted the SAJ HACS, removed all data from YAML regarding the SAJ, reinitialized HASS, reinstalled SAJ HACS, reinitialized HASS, added SAJ yaml script and reinitialized HASS.

Still received the same error.

See attached yaml without pass and user :) and I have also included log file swell as files within folder custom_componets/saj_esolar

Thanks again for your help.

A￼￼￼￼￼￼

djansen1987, 2024-08-09:
No response

closed 2024-08-09 by djansen1987