djansen1987/SAJeSolar #86: Unvaiable same sensor after update

1achy, 2024-08-22:
[image]
[image]

After about 12.00 same sensor unvaiable, can i check?

djansen1987, 2024-08-22:
What does the logs say?

1achy, 2024-08-23:
2024-08-23 07:42:34.768 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: nowPower State: 411.0
2024-08-23 07:42:34.769 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: runningState State: Yes
2024-08-23 07:42:34.769 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: devOnlineNum State: Yes
2024-08-23 07:42:34.770 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: todayElectricity State: 0.25
2024-08-23 07:42:34.770 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: monthElectricity State: 730.45
2024-08-23 07:42:34.771 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: yearElectricity State: 1536.7
2024-08-23 07:42:34.771 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalElectricity State: 4355.34
2024-08-23 07:42:34.772 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfUseRate State: 59.75%
2024-08-23 07:42:34.772 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalBuyElec State: 880.47
2024-08-23 07:42:34.773 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalConsumpElec State: 3415.23
2024-08-23 07:42:34.774 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalSellElec State: 1752.85
2024-08-23 07:42:34.775 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: lastUploadTime State: 2024-08-23 06:36:23
2024-08-23 07:42:34.776 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalPlantTreeNum State: 7.72
2024-08-23 07:42:34.776 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalReduceCo2 State: 4.34
2024-08-23 07:42:34.777 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: plantuid State: xxxxxx
2024-08-23 07:42:34.777 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: currency State: $
2024-08-23 07:42:34.778 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: address State: xxxx
2024-08-23 07:42:34.778 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: isOnline State: Y
2024-08-23 07:42:34.778 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: status State: success
2024-08-23 07:42:34.779 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalLoadPower State: 411.0
2024-08-23 07:42:34.779 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: batCapcity State: 100.0
2024-08-23 07:42:34.780 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: batCurr State: -0.72
2024-08-23 07:42:34.781 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: batEnergyPercent State: 21.0
2024-08-23 07:42:34.781 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: batteryDirection State: Standby
2024-08-23 07:42:34.782 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: batteryPower State: 0.0
2024-08-23 07:42:34.782 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: gridPower State: 0.0
2024-08-23 07:42:34.783 ERROR (MainThread) [custom_components.saj_esolar.sensor] Grid Direction unknown value: 0
2024-08-23 07:42:34.783 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: gridDirection State: 0
2024-08-23 07:42:34.783 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: h1Online State: Yes
2024-08-23 07:42:34.784 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: outPower State: 411.0
2024-08-23 07:42:34.784 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: outPutDirection State: Exporting
2024-08-23 07:42:34.784 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: pvPower State: 411.0
2024-08-23 07:42:34.785 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: solarPower State: 6.0

2024-08-23 07:43:35.729 DEBUG (MainThread) [custom_components.saj_esolar.sensor] {'storeDevicePower': {'type': 1, 'pvPower': 427.0, 'gridPower': 0.0, 'batteryPower': 0.0, 'outPower': 427.0, 'totalLoadPower': 427.0, 'solarPower': 6.0, 'batCurr': -0.72, 'batEnergyPercent': 21.0, 'runningState': 1, 'isOnline': 1, 'isAlarm': 0, 'mark': 1, 'batCapcity': 100.0, 'hasMeter': True, 'hasBattery': True, 'batCapcityStr': '100.00Ah', 'pvDirection': 1, 'gridDirection': 0, 'batteryDirection': 0, 'outPutDirection': 1, 'hFaultMSG': 0, 'dataTime': 1724366400000, 'updateDate': 1724391674000, 'buyPower': 0, 'sellPower': 0, 'chgPower': 0, 'dischgPower': 0}, 'type': 1, 'status': 'success'}
2024-08-23 07:43:35.730 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Set-Cookie:

cirinx, 2024-08-23:
same problem

flauteps, 2024-08-26:
Same problem

manducagiuseppe, 2024-08-30:
Same problem

djansen1987, 2024-08-30:
@jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?

manducagiuseppe, 2024-08-30:
> @jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?
even if I install an old version some entities don't work anymore

sensor.esolar_pvelec
sensor.esolar_useelec
sensor.esolar_buyelec
sensor.esolar_sellelec
sensor.esolar_buyrate
sensor.esolar_sellrate
sensor.esolar_selfconsumedrate1
sensor.esolar_selfconsumedrate2
sensor.esolar_selfconsumedenergy1
sensor.esolar_selfconsumedenergy2
sensor.esolar_isalarm
sensor.esolar_chargeelec
sensor.esolar_dischargeelec

djansen1987, 2024-08-30:
Duplicate issue https://github.com/djansen1987/SAJeSolar/issues/87

Conversation continues in this one.

jaesbit, 2024-08-30:
> @jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?

I'll take a look to this issue on this weekend

walkjivefly, 2024-08-30:
I am running slightly modded SAJ eSolar https://github.com/faanskit/ha-esolar (it didn't work for me originally (July 2023) but no-one else was complaining about the same issues so I just kludged it to work for me and didn't open a PR). Last weekend it stopped working with a "peakPower" error. It turned out SAJ have changed the plant type for systems with batteries from 3 to 1. (iirc they changed it the other way some months ago.) 

So, for me the fix was to change the test for 
```
    if plant["type"] == 3:
```
to 
```
    if plant["type"] == 1:
```
around line 229 in esolar.py.

It's so long since I set this up, and my memory is so bad, I had gotten mixed up about which SAJ integration I am using. Hopefully this hint can help here as well but given there is no esolar.py in this integration the change will need to happen elsewhere.

djansen1987, 2024-09-01:
> > @jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?
> 
> I'll take a look to this issue on this weekend

Were you able to work on a fix?

jaesbit, 2024-09-01:
> > > @jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?
> > 
> > 
> > I'll take a look to this issue on this weekend
> 
> Were you able to work on a fix?

I'm on it, I just start few minutes ago. I hope that it will be a fast fix

jaesbit, 2024-09-01:
> Thank you!

I see, the issue, the viewBean, from getPlatnDetailChart2 is not in the response, but i see it on the request that they (saj-electric) made in their API. I will review the arguments of the request maybe some one are missing. 

More info.... I executed the request in the browser context and no data is returned. But if I fill the `elecDevicesn` parameter in the request all `viewBean` is filled. But, I need to check what values will be returned if no Battery are in the plant

I'll be on discord, if you wan to connect directly in real time.

jaesbit, 2024-09-01:
More info: I can use both (Battery or inverter) as value for `deviceSnArr` but for `elecDevicesn` only it will work with Battery SN. I think it was an issue related to an upgrade of their api by requiring this value to be filled in the request

[image]

djansen1987, 2024-09-01:
It is 10.30 here so won't be able to join you on discord and going to bed 😉 als I don't have a say system anymore my self. Maybe you can get data from previous post,they have shared some json if I do not mistake.

djansen1987, 2024-09-01:
Created a new release (and now really going to bed 😉)

cbaeyens, 2024-09-02:
Hi,

I can confirm that everything is working again. All entities have values and they are being updated.
Some screenshots of some entities attached.
[image]
[image]

A big thank you to everyone who contributed to the solution!

Chris

djansen1987, 2024-09-02:
Thank you for confirming chris!

1achy, 2024-09-02:
After FIX all work thanks very thanks

closed 2024-10-19 by djansen1987