djansen1987/SAJeSolar #87: Issues reading battery data.

reneneumann, 2024-08-25:
Hi, 

I'm using this plugin for a while without major issues but since day 1 do I have an issue getting my battery data gathered by home assistant. The log displayes the following lines:

```
2024-08-25 19:50:46.809 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: isOnline State: Y
2024-08-25 19:50:46.809 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: status State: success
2024-08-25 19:50:46.809 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: systemPower State: 5.0
2024-08-25 19:50:46.809 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: sellElec State: None
2024-08-25 19:50:46.809 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: buyRate State: None
2024-08-25 19:50:46.810 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: sellRate State: None
2024-08-25 19:50:46.810 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfConsumedRate1 State: None
2024-08-25 19:50:46.810 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfConsumedRate2 State: None
2024-08-25 19:50:46.829 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfConsumedEnergy1 State: None
2024-08-25 19:50:46.830 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfConsumedEnergy2 State: None
2024-08-25 19:50:46.830 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: chargeElec State: None
2024-08-25 19:50:46.830 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: dischargeElec State: None
2024-08-25 19:51:16.805 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: nowPower State: 271.0
2024-08-25 19:51:16.806 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: runningState State: Yes
2024-08-25 19:51:16.806 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: devOnlineNum State: Yes
2024-08-25 19:51:16.806 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: todayElectricity State: 11.33
2024-08-25 19:51:16.807 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: monthElectricity State: 536.97
2024-08-25 19:51:16.807 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: yearElectricity State: 3162.03
2024-08-25 19:51:16.807 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalElectricity State: 3162.86
2024-08-25 19:51:16.807 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: selfUseRate State: 73.99%
2024-08-25 19:51:16.807 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalBuyElec State: 805.64
2024-08-25 19:51:16.808 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalConsumpElec State: 3157.4
2024-08-25 19:51:16.808 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalSellElec State: 822.81
2024-08-25 19:51:16.808 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: lastUploadTime State: 2024-08-25 19:44:30
2024-08-25 19:51:16.808 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalPlantTreeNum State: 5.61
2024-08-25 19:51:16.808 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: totalReduceCo2 State: 3.15
```

I tried for the repo the manual test python script and this reported the values for:
chargeElec
dischargeElec

Below is an example:

```
{
  "pageNo": 1,
  "pageSize": 10,
  "plantList": [
    {
      "plantuid": "CFA...F44",
      "plantname": "",
      "systempower": 5.0,
      "type": 1,
      "installername": "",
      "countryCode": "",
      "country": "Ireland",
      "province": "",
      "city": "",
      "county": "",
      "foreignRemark": "",
      "address": ".....SAJ B2 5kWh",
      "latitude": 0.0,
      "longitude": 0.0,
      "createDateStr": "2024-08-24 13:37:16",
      "isOnline": "Y",
      "runningState": 1,
      "nowPower": 498.0,
      "todayElectricity": 10.21,
      "totalElectricity": 2708.7,
      "enableEdit": "Y",
      "enableDelete": "Y",
      "enableVisitor": "Y",
      "isFavorite": "0",
      "isRename": 0,
[... 283 lines cut ...]
            "totalLoadPowerWatt": 1.0,
            "todaySelfConsumpEnergy": 0.0,
            "rOutVoltStr": "240.2",
            "todayLoadEnergyStr": "23.41",
            "rGridPowerWatt": 494.0,
            "sOutCurr": 0.0,
            "pV3StrCurr4Str": "N/A",
            "sGridFreq": 0.0,
            "tGridPowerVA": 0.0,
            "meterAStatus": 0,
            "pV1Volt": 335.8,
            "sOnGridOutVolt": 0.0,
            "todaySelfConsumpEnergyStr": "0",
            "plantName": ""
          }
        }
      ]
    }
  ],
  "pageHtml": "<ul>\n<li class=\"disabled\"><a href=\"javascript:;\">\u00ab Previous</a></li>\n<li class=\"active\"><a href=\"javascript:;\">1</a></li>\n<li class=\"disabled\"><a href=\"javascript:;\">Next \u00bb</a></li>\n<li class=\"disabled controls pageC\">Page <input type=\"text\" value=\"1\" onkeypress=\"var e=window.event||event;var c=e.keyCode||e.which;if(c==13)page(this.value,10,'');\" onclick=\"this.select();\"/> Total 1 Records</ul>\n<div style=\"clear:both;\"></div>",
  "status": "success"
}
```

Any idea what I'm potentially wrong in the configuration?

acasandreiflorin, 2024-08-26:
Hi, I have a similar issue, after the latest update the chargeElec and dischargeElec are not reported anymore, the log shows the below:
saj_esolar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 727, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1300, in async_device_update
    await self.async_update()
  File "/config/custom_components/saj_esolar/sensor.py", line 1019, in async_update
    if 'dischargeElec' in energy['viewBean']:
                          ~~~~~~^^^^^^^^^^^^
KeyError: 'viewBean'

saj_esolar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 727, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1300, in async_device_update
    await self.async_update()
  File "/config/custom_components/saj_esolar/sensor.py", line 1015, in async_update
    if 'chargeElec' in energy['viewBean']:
                       ~~~~~~^^^^^^^^^^^^
KeyError: 'viewBean'

cbaeyens, 2024-08-30:
I have the same issue

djansen1987, 2024-08-30:
@jaesbit it seems like your pull request broke the some bits for these user, are you able to fix it or should we revert the you update?

marcosturiale, 2024-08-30:
I have the same issue. 
Here there are the complete list of the resources that currently no longer return data after this update.

      - chargeElec 
      - dischargeElec
      - pvElec
      - useElec
      - buyElec
      - sellElec
      - buyRate
      - sellRate
      - selfConsumedRate1
      - selfConsumedRate2
      - selfConsumedEnergy1
      - selfConsumedEnergy2 

My Inverter is H1.

Thank you for your support.

djansen1987, 2024-08-30:
Issue continuous in first report

https://github.com/djansen1987/SAJeSolar/issues/86

closed 2024-08-30 by djansen1987