djansen1987/SAJeSolar #90: Unavailable sensor.esolar_peakpower

gusmori, 2024-09-13:
Since the last update of Home Assistant the sensor "sensor.esolar_peakpower" appears as unavailable.
Is there any way to recover it without having to restore a backup?
Regards.

gusmori, 2024-09-13:
It was from an update I made to the integration on September 2nd. I don't know how to get the log. Sorry.

cmgestevao, 2024-10-17:
> It was from an update I made to the integration on September 2nd. I don't know how to get the log. Sorry.

To get the logs:

Debugging
Add the relevant lines below to the configuration.yaml:

```
logger:
  default: info
  logs:
    custom_components.saj_esolar: debug
```

When i get mine will post them here, yesterday i have updated to version 1.4.2 and also lost the peak_power value, the sensor its unknow!

cmgestevao, 2024-10-17:
Update:

Log with version 1.4.2:
### 2024-10-17 20:55:30.225 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: peakPower State: None

Log with version 1.4.0:
### 2024-10-17 20:36:53.504 DEBUG (MainThread) [custom_components.saj_esolar.sensor] Device: peakPower State: 1680.0

If I replace this line in last version:
`  url4 = f"{self._provider.getBaseUrl()}/monitor/site/getPlantDetailChart2?plantuid={plantuid}&chartDateType=1&energyType=0&clientDate={clientDate}&deviceSnArr={deviceSnArr}&chartCountType=2&previousChartDay={previousChartDay}&nextChartDay={nextChartDay}&chartDay={chartDay}&previousChartMonth={previousChartMonth}&nextChartMonth={nextChartMonth}&chartMonth={chartMonth}&previousChartYear={previousChartYear}&nextChartYear={nextChartYear}&chartYear={chartYear}&elecDevicesn={deviceSnArr}&_={epochmilliseconds}"`

with the line from version 1.4.0:
`  url4 = f"{self._provider.getBaseUrl()}/monitor/site/getPlantDetailChart2?plantuid={plantuid}&chartDateType=1&energyType=0&clientDate={clientDate}&deviceSnArr={deviceSnArr}&chartCountType=2&previousChartDay={previousChartDay}&nextChartDay={nextChartDay}&chartDay={chartDay}&previousChartMonth={previousChartMonth}&nextChartMonth={nextChartMonth}&chartMonth={chartMonth}&previousChartYear={previousChartYear}&nextChartYear={nextChartYear}&chartYear={chartYear}&elecDevicesn=&_={epochmilliseconds}"`

the peakpower sensor returns the right value from the plant!

 
Hope this can help the developers to understand what is happening in last version (1.4.2)

My devices are:
R5-1.5K-S1-15
SEC Module

djansen1987, 2024-10-18:
@jaesbit 

Would you perhaps know why this has stopped working for @gusmori after https://github.com/djansen1987/SAJeSolar/pull/88 ?

djansen1987, 2024-10-18:
@gusmori could you please share us you configuration.yaml esolar part (without credentials)

jaesbit, 2024-10-18:
> @jaesbit
> 
> Would you perhaps know why this has stopped working for @gusmori after #88 ?

I'm not sure, because in my installation have a Battery, and I don't have SEC module. But seems like without battery or with SEC module the `elecDevicesn={deviceSnArr}` needs to be empty a good test could be trying to add there de SEC module Sn.

cmgestevao, 2024-10-18:
> @gusmori could you please share us you configuration.yaml esolar part (without credentials)

My configuration.yaml is:

```
- platform: saj_esolar
  username: !secret saj_esolar_user
  password: !secret saj_esolar_pass
  plant_id: 0 # Default is 0. Typically ordered in the same way as they are listed in the eSolar app
  sensors: saj_sec # Optional will only work with SAJ Sec Module
  resources:
    - nowPower
    - runningState
    - todayElectricity
    - monthElectricity
    - yearElectricity
    - totalElectricity # Energy -> Solar production
    - todayGridIncome
    - income
    - lastUploadTime
    - totalPlantTreeNum
    - totalReduceCo2
    - status
    - plantuid
    - currency
    - address
    - isOnline
    - peakPower
    - systemPower # Installed capacity

    - pvElec
    - useElec
    - buyElec
    - sellElec
    - buyRate
    - sellRate
    - selfConsumedRate1
    - selfConsumedRate2
    - selfConsumedEnergy1
    - selfConsumedEnergy2
    - plantTreeNum
    - reduceCo2
    - totalGridPower
    - totalLoadPower
    - totalPvgenPower
    - totalPvEnergy
    - totalLoadEnergy # Energy -> Grid consumption
    - totalBuyEnergy
    - totalSellEnergy # Energy -> Return to grid
    #these entities are deprecated since they return incorrect values
    - totalGridPower # Power being exported to the grid
    - totalLoadPower # Solar power being currently self-consumed
    - totalPvgenPower # Power imported from the grid
    #these new entities replace them
    - gridLoadPower # Power imported from the grid
    - solarLoadPower # Solar power being currently self-consumed
    - homeLoadPower # Total power being consumed by the plant (the home)
    - exportPower # Power being exported to the grid
```

I have tried to add the serial number in that line like this:

`elecDevicesn={M53XXXXXXXXXXXXX}` 

[image]

with this change all sensors are unknow, if I leave "elecDevicesn" empty all sensors are ok!

jaesbit, 2024-10-18:
@cmgestevao which Serial number number did you add? the Inverter or Sec module. And when you add de serial please avoid tu use brakets (are for variables) so use `...&elecDevicesn=1234567890&_={....` with your sec SN. If I'm not wrong the SN that is used by default is de inverter one.

Also it could be interesting to this library to have de SEC module TYPE ID (I supposed is `1`) but you can confirm it by adding [here](https://github.com/djansen1987/SAJeSolar/blob/main/custom_components/saj_esolar/sensor.py#L703)

`_LOGGER.info(f"devices SN {', '.join(f'{item['devicesn']}:{item['type']}' for item in plantDetails['list'])}")` 

Thanks

cmgestevao, 2024-10-18:
> @cmgestevao which Serial number number did you add? the Inverter or Sec module. And when you add de serial please avoid tu use brakets (are for variables) so use [...]

I have tried both.

Doing more testing, the update is like this:

"elecdevicesn" empty:

[image]

The result is:

[image]

"elecdevicesn" with SN of inverter:

[image]

The result is:

[image]

"elecdevicesn" with SN of SEC:

[image]

The result is:

[image]

"Desconhecido" means unknow

Thanks

jaesbit, 2024-10-18:
@djansen1987 [here](https://github.com/djansen1987/SAJeSolar/blob/main/custom_components/saj_esolar/sensor.py#L728) we need to remove SN if h1 is not in the config `&elecDevicesn={deviceSnArr}&_` so something like this `&elecDevicesn={deviceSnArr if self.sensors == 'h1' else ''}&_` in one line or 

```
elecDevicesn = deviceSnArr if self.sensors == "h1" else ""
url4 = f"....&elecDevicesn={elecDevicesn}&_...."
```

I can't create PR until sunday, if you can I think this should work

cmgestevao, 2024-10-18:
> @djansen1987 [here](https://github.com/djansen1987/SAJeSolar/blob/main/custom_components/saj_esolar/sensor.py#L728) we need to remove SN if h1 is not in the config `&elecDevicesn={deviceSnArr}&_` so [...]

Tested this and it's ok!
```
elecDevicesn = deviceSnArr if self.sensors == "h1" else ""
url4 = f"....&elecDevicesn={elecDevicesn}&_...."

```

[image]

djansen1987, 2024-10-18:
Are you able to create a pull request?

cmgestevao, 2024-10-18:
Sorry, don't know how

cmgestevao, 2024-10-18:
If there is something wrong, please let me know, this is my first PR!

djansen1987, 2024-10-19:
Yes looks great, will merge it

djansen1987, 2024-10-19:
released in https://github.com/djansen1987/SAJeSolar/releases/tag/v1.5.7

closed 2024-10-19 by djansen1987