djansen1987/SAJeSolar #91: unknown

mrpontes, 2024-09-17:
Olá gostaria de uma ajuda para entender o motivo pelo qual a minha integração com o Home Assistant não está funcionando.

[image]

Meus sensores ficam todos com esta status.

Se abro pelo o site vejo que está funcionando

[image]

[image]

djansen1987, 2024-09-17:
What does the logs say? (Hide sensitive information)

mrpontes, 2024-09-17:
[image]

mrpontes, 2024-09-17:
> O que os logs dizem? (Ocultar informações provisórias)

Este erro originou-se de uma integração personalizada.

Registrador: custom_components.saj_esolar.sensor
Fonte: custom_components/saj_esolar/sensor.py:838
integração: saj_esolar (documentação, problemas)
Ocorreu pela primeira vez: 12:41:25 (22 ocorrências)
Último registro: 14:26:26

Cannot poll eSolar using url: %s

djansen1987, 2024-09-18:
What brand do you have ? Might be that you have to use a different domainname

djansen1987, 2024-09-18:
Difficult to determine remotely, dubbel checked username password, did you may configure multifactor authentication?

mrpontes, 2024-09-18:
> Difficult to determine remotely, dubbel checked username password, did you may configure multifactor authentication?

A senha e o e-mail eu já confirmei, está certo agora não sei onde fica essa autenticação multifator

djansen1987, 2024-09-19:
Did it work before or is this the first time setting up the integration? The error says it cannot correctly login. This mostly can mean a few things 

- bad username or password
- multifactor is enabled
-  you use a different portal url then the default
- not installed the latest version of the integration
- you account is an installer/reseller account
- you have multiple plants and need to select the correct one.

Other then that i don't know. You can also enable debug mode and look in de full logs to see if you can spot something wrong.

mrpontes, 2024-09-20:
> Did it work before or is this the first time setting up the integration? The error says it cannot correctly login. This mostly can mean a few things
> 
> * bad username or password 
      - **usuário e senha está correta, consigo ver o funcionamento via web e app**
> * multifactor is enabled
     -  **Não sei onde habilita ou desabilita essa opção de multifactor.**
> * you use a different portal url then the default
     - **Uso essa url (https://intop.saj-electric.com/saj)**
> * not installed the latest version of the integration
    - **A integração uso a última disponível no Home Assistant.**
> * you account is an installer/reseller account
   - **Minha conta é de cliente final**
> * you have multiple plants and need to select the correct one.
   - **Uso somente uma planta**
> 
> Other then that i don't know. You can also enable debug mode and look in de full logs to see if you can spot something wrong.
  - **Poderia me orientar como habilitar o modo de depuração**

_Obrigado desde já por seu empenho em querer me ajudar.  
Respostas acima nos questionamento._

mrpontes, 2024-09-25:
Desculpe-me acabei fechando por engano, ainda continuo com o problema.  Estava olhando o log do Home Assistant, tem esse erro.  Veja se consegue entender 
[image]

[image]

mrpontes, 2024-10-10:
Boa tarde, voltando nesse assunto
Esse erro acima, diz para você alguma coisa?

djansen1987, 2024-10-18:
You can enable the debug log by updating your configuration.yaml like below:

`logger:
  default: info
  logs:
    custom_components.saj_esolar: debug`

mrpontes, 2024-10-18:
> Você pode habilitar o log de depuração atualizando seu configuration.yaml como abaixo:
> 
> ```yaml
> logger:
>   default: info
>   logs:
>     custom_components.saj_esolar: debug
> ```

Obrigado pelo retorno, essa linha de depuração já estava no meu configuration.yaml

djansen1987, 2024-10-18:
Difficult to say, could you post the saj_esolar part of your configuration.yaml. make sure no password or username is posted on github!!

mrpontes, 2024-10-18:
> Difficult to say, could you post the saj_esolar part of your configuration.yaml. make sure no password or username is posted on github!!

[image]

djansen1987, 2024-10-19:
English:

The configuration appears to be correct. Unfortunately, without access to your eSolar login details, it's difficult to pinpoint the exact issue.

However, if you're comfortable sharing your login credentials with me, you can send them securely to github@djansen.nl. I'll then try to set it up in my development environment and see if I can get it working for you.

Of course, I completely understand if you're hesitant to share this information.

Português (Brasil):

A configuração parece estar correta. Infelizmente, sem dar uma olhada nas suas credenciais de login do eSolar, é difícil identificar o problema exato.

No entanto, se você se sentir confortável em compartilhar seus dados de login comigo, pode enviá-los com segurança para github@djansen.nl. Vou tentar configurá-lo no meu ambiente de desenvolvimento e ver se consigo fazê-lo funcionar para você.

É claro, eu entendo completamente se você hesitar em compartilhar essas informações.

mrpontes, 2024-10-19:
> English:
> 
> The configuration appears to be correct. Unfortunately, without access to your eSolar login details, it's difficult to pinpoint the exact issue.
> 
> However, if you're comfortable sharing your login credentials with me, you can send them securely to [github@djansen.nl](mailto:github@djansen.nl). I'll then try to set it up in my development environment and see if I can get it working for you.
> 
> Of course, I completely understand if you're hesitant to share this information.
> 
> Português (Brasil):
> 
> A configuração parece estar correta. Infelizmente, sem dar uma olhada nas suas credenciais de login do eSolar, é difícil identificar o problema exato.
> 
> No entanto, se você se sentir confortável em compartilhar seus dados de login comigo, pode enviá-los com segurança para [github@djansen.nl](mailto:github@djansen.nl). Vou tentar configurá-lo no meu ambiente de desenvolvimento e ver se consigo fazê-lo funcionar para você.
> 
> É claro, eu entendo completamente se você hesitar em compartilhar essas informações.

Muito obrigado por seu empenho em querer me ajudar, fico muito feliz e vou sim sem problemas te enviar por e-mail as minhas credenciais.

English:

Thank you very much for your commitment in wanting to help me, I am very happy and I will send you my credentials by email without any problems.

djansen1987, 2024-10-19:
English:

I found the issue! If I had read more carefully, I would have known. You're using the international website of SAJ, not the default URL. Therefore, you should add the following extra line to your configuration:

`provider_domain: intop.saj-electric.com`

The complete configuration should then look like this:

```YAML
- platform: saj_esolar
  username: Username  # (Replace with your actual username)
  password: password  # (Replace with your actual password)
  provider_domain: intop.saj-electric.com
  plant_id: 0         # Default is 0. Typically ordered as listed in the eSolar app
  resources:
    - nowPower
    - runningState
    - todayElectricity
    - monthElectricity
    - yearElectricity
    - totalElectricity  # Energy -> Solar production
    - todayGridIncome
    - income
    - lastUploadTime
    - totalPlantTreeNum
    - totalReduceCo2
    - status
    - plantuid
    - currency
    - address
    - isOnline
    - peakPower
    - systemPower       # Installed capacity
```

Português (Brasileiro):

Encontrei o problema! Se eu tivesse lido a documentação com mais cuidado, teria percebido. Você está usando o site internacional da SAJ, e não a URL padrão. Portanto, você deve adicionar a seguinte linha extra à sua configuração:

`provider_domain: intop.saj-electric.com`

A configuração completa deve ficar assim:

```YAML
- platform: saj_esolar
  username: Nome de usuário  # (Substitua pelo seu nome de usuário real)
  password: senha         # (Substitua pela sua senha real)
  provider_domain: intop.saj-electric.com
  plant_id: 0             # O padrão é 0. Normalmente ordenado como listado no aplicativo eSolar
  resources:
    - nowPower
    - runningState
    - todayElectricity
    - monthElectricity
    - yearElectricity
    - totalElectricity  # Energia -> Produção solar
    - todayGridIncome
    - income
    - lastUploadTime
    - totalPlantTreeNum
    - totalReduceCo2
    - status
    - plantuid
    - currency
    - address
    - isOnline
    - peakPower
    - systemPower       # Capacidade instalada
```

djansen1987, 2024-10-19:
English:

Be aware that the international portal also has a new platform button. Do not move to the new platform, otherwise this integration won't work anymore because of multifactor authentication.

Portuguese (Brazilian):

Fique atento que o portal internacional também possui um novo botão de plataforma. Não mude para a nova plataforma, caso contrário, essa integração não funcionará mais devido à autenticação multifator.

[image]

mrpontes, 2024-10-19:
> English:
>
> Be aware that the international portal also has a new platform button. Do not move to the new platform, otherwise this integration won't work anymore because of multifactor [...]

Meu amigo funcionou perfeito agora!
Obrigado pela atenção que deu ao meu caso!
Deus abençoe!

djansen1987, 2024-10-20:
Good to here, thank you for your response

closed 2024-10-20 by djansen1987