djansen1987/SAJeSolar #92: After inverter change, cannot get all values

nenewapete, 2024-10-02:
Hi,

greenheiss changed my inverter under warranty because of hw issues. Also, the batteries.
After that, on the greenheiss portal I can see everything is ok, but on HA, no changed parameters, but only can retrieve Solar production. Not Buy or sell or batteries chrg/dschrg...

`sensor.esolar_totalbuyelec (unavailable)
`
Every parameter shows like this, except for the "esolar totalElectricity"

Can you help me?

djansen1987, 2024-10-18:
Can you enable debug and show the logs?

```yaml
logger: default: info logs: custom_components.saj_esolar: debug
```

djansen1987, 2024-10-20:
Maybe it is because you now have 2 plants in your tenant and you have to select the second plant id

santiagozky, 2025-09-28:
closing for lack of activity

closed 2025-09-28 by santiagozky