djansen1987/SAJeSolar #101: I would like to obtain more battery data.

mactron254, 2024-11-21:
I'm using your wonderful integration and I would like to know if there is any way to obtain battery data such as battery temperature, health status, current capacity in kW, or the remaining charge and discharge time.

I'm currently using the H1 battery with the new version of the Elekeeper app, which allows me to see all this information. I would like to be able to view it from Home Assistant.

Thank you for your great work

djansen1987, 2024-11-30:
Hi @mactron254,

Thank you for you post! Currently, this data isn't directly supported by the module.

However, if the website exposes an API endpoint for this specific data, we may be able to integrate it.

To determine feasibility, please check the following:

Identify the API Endpoint:

Use your browser's developer tools (Network tab) to inspect the network requests made when the data is displayed on the website.
Look for the URL of the API endpoint that fetches the required data.
Analyze the Request:

In the same Network tab, examine the Request headers and body.
This will provide information on the necessary authentication, parameters, and payload format.
Once we have this information, we can assess the effort required to incorporate it into the module.

Please let me know if you have any further questions or require more specific guidance.

mrjuvas, 2025-01-28:
Hi @djansen1987 
Thanks for your great work with this integration!
I think I found at least some of the information you requested above.

Seems like the API Endpoint (or at least one example API) is "/saj/cloudMonitor/deviceInfo/findBatteryRealTimeList".

This endpoint is used on the Battery realtime data page: 
"https://fop.saj-electric.com/saj/cloudMonitor/deviceInfo/toACBatteryRealTimeData?devicesn=SERIALNUMBEROFTHEINVERTER"

Example of ajax call:
dd.ajax("/saj/cloudMonitor/deviceInfo/findBatteryRealTimeList", {
				data : { devicesn:'SERIALNUMBEROFTHEINVERTER', timeStr:timeStr
// 					timeStart:dateStr+" "+timeStart,
// 					timeEnd:dateStr+" "+timeEnd
				},

[image]