djansen1987/SAJeSolar #102: Glitch in buyElec around 12 AM

rinyakok, 2024-12-22:
Hello,

I'm using your amazing component to read my inverter data in HA.
I realized that there is a glitch in buyElec sensor data around 12 AM every day.
After midnight, the sensor value set back to zero for one read, after that it jumps back and continuos increasing from previous day maximum value for one read, then it gets back to around 0. (See attached log)

sensor.esolar_buyelec,4.33,2024-12-20T23:00:00.000Z
sensor.esolar_buyelec,0.0,2024-12-20T23:01:10.682Z
sensor.esolar_buyelec,4.34,2024-12-20T23:03:40.691Z
sensor.esolar_buyelec,0.01,2024-12-20T23:08:40.704Z
...
sensor.esolar_buyelec,8.93,2024-12-21T22:48:45.157Z
sensor.esolar_buyelec,0.0,2024-12-21T23:01:15.178Z
sensor.esolar_buyelec,8.94,2024-12-21T23:03:45.183Z
sensor.esolar_buyelec,0.0,2024-12-21T23:08:45.192Z

It causes a peak on my energy dashboard and wrong calculation of my energy usage & self sufficient.

I use the version 15.6. where I changed the refresh interval to 150 seconds.

What could be wrong ? How to fix ?

Thank you ,
[history.csv](https://github.com/user-attachments/files/18222753/history.csv)
[image]
[image]

djansen1987, 2025-01-10:
it seems you are the only one at the moment, do you maybe shutdown your home assistant at night ?

rinyakok, 2025-01-14:
No. It runs continuously.

odjuvsla, 2025-01-21:
Unfortunately your not the only one. I had the same issue, and now I started getting spurious spikes at other random points in time as well. I logged the response from the SAJ solar API, and the erroneous data come from the API.

santiagozky, 2025-04-20:
unfortunately most of the problems come from the platform itself, not this integration.
Those spikes normally come when at some point the platform returns an incorrect value (lower than the previous one), so the next time, when the correct value come, HA will record a larger than usual consumption.

e.g. for this series of data points
1- 1000kwh (correct)
2- 1001 kwh (correct): HA stores 1kwh of consumption
3-  0 kwh (incorrect); HA stores 0kwh consumption
(more incorrect zeros might come here)
n- 1003kwh (correct): HA stores 1003kwh instead of 2. you get that huge spike

I normally just go to statistics, find the correcty entity and click the "outliers" button. that usually finds the correct data point to fix.

Im not sure this integration can do much about this.

I've set an automation to let me know when any of the integration's entities have a value of zero or less so I know I have to fix it