djansen1987/SAJeSolar #103: inversores-style.greenheiss.com

javmorarr, 2025-01-09:
Would it be possible to capture the battery data? I have checked the portal and found this:
[image]

djansen1987, 2025-01-09:
Try to configure the H1 module following the documentation

javmorarr, 2025-01-09:
Battery and mains Gridpower data are not loaded.
[image]

Gallach, 2025-03-03:
Hi,
I also have a Greenheiss inverter and I can only get the produced power but I am also missing the battery and grid info, there is any update regarding this?
Thanks!

santiagozky, 2025-09-28:
@Gallach  @javmorarr  if you are still using this integration. maybe you can check the 1.6.0 beta? I could use some help from someone with an H1

javmorarr, 2025-09-29:
Ok la instalo y os informo.Gracias Enviado desde mi iPhone

santiagozky, 2025-11-28:
closing because inactivity

closed 2025-11-28 by santiagozky