djansen1987/SAJeSolar #104: iop.saj-electric.com/saj/login returned 405

HardSource, 2025-02-09:
I'm trying to configure it for the `saj-electric.com` domain on the international node, but the logs return error 405.

```
2025-02-09 14:35:39.700 ERROR (MainThread) [custom_components.saj_esolar.sensor] https://iop.saj-electric.com/saj/login returned 405
2025-02-09 14:35:39.702 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
2025-02-09 14:35:39.702 ERROR (MainThread) [custom_components.saj_esolar.sensor] return data NONE
...
``` 

I'm using this configuration:
```
sensor:
  - platform: saj_esolar
    username: XXXX@XXXX.com
    password: XXXX
    provider_domain: iop.saj-electric.com
    resources:
      - nowPower
      - runningState
      - todayElectricity
      - monthElectricity
      - yearElectricity
      - totalElectricity
      - todayGridIncome
      - income
      - lastUploadTime
      - totalPlantTreeNum
      - totalReduceCo2
      - isAlarm
      - status
      - plantuid
      - currency
      - address
      - isOnline
      - peakPower
      - systemPower
 ``` 

It would be great if the integration worked for this domain, I can help with anything you need.

Thanks in advance.

djansen1987, 2025-02-10:
When you login to the webportal do you need to solve a captcha first or have multifactor configured? If that is the case, it will not work unfortunately. This is then blocked by the provider

HardSource, 2025-02-12:
I don't need a captcha and I'm not using two-factor authentication. I log in with my email and password credentials only.

santiagozky, 2025-04-20:
isn't that domain the new platform that SAJ is introducing? IIUC this integration wont work with the new platform only with https://fop.saj-electric.com and their whitelabel reskins.

closed 2025-09-28 by santiagozky