djansen1987/SAJeSolar #108: future of this integration (SAJ vs rebranded)

santiagozky, 2025-04-22:
Hello,

It seems that SAJ will very soon move their clients to their new platform, which means that this integration will not be useful for SAJ clients without a change. It does also seems that most reseller are not moving their clients to this new platform (I have no notice from greenheiss at all and while my credentials work in the new SAJ system, I get no data in them).

So this begs the question about the future of this integration. This integration, as it stands now, will most likely be still be useful for people with rebranded inverters (which will stay in their rebranded platforms) but not for people on SAJ branded inverters. and if it is reworked for the new platform, it will break for rebranded ones.

@djansen1987 Do you plan to continue developing this or acepting PRs for this integration? I see some possible options:

- You stop working on this (for which I would not blame you at all, since you dont have any personal stake in this anymore) and users of each respective platform keep working on a integration for each case through a fork or whatever.
- This integrations keeps supporting the existing platform and a new integration is created for the new one. 
- This integration moves to the new platform and a fork is created to maintain the existing one.
- This integration somehow supports both? (seems unlikely if the platforms are too different.)

I think this is more pression for SAJ users since this will break for them soon, but it would be unfortunate if a PR for the new platform comes and breaks stuff for rebranded users.

djansen1987, 2025-04-22:
Hi @santiagozky ,

I do accept pr's also if you like you can become a owner of the repo. I don't have an saj device my self anymore. My uncle still has, and we are trying the rs485 port option for him.

jjmmbb, 2025-04-22:
> Hi [@santiagozky](https://github.com/santiagozky) ,
> 
> I do accept pr's also if you like you can become a owner of the repo. I don't have an saj device my self anymore. My uncle still has, and we are trying the rs485 port option for him.

rs485 will need a device to convert to WIFI, right? What device are you using?

jjmmbb, 2025-04-22:
> Hello,
>
> It seems that SAJ will very soon move their clients to their new platform, which means that this integration will not be useful for SAJ clients without a change. It does also seems that most [...]

Have you tried this one: https://github.com/faanskit/ha-esolar ?

djansen1987, 2025-04-22:
> Have you tried this one: https://github.com/faanskit/ha-esolar ?

Yes, actually this integration is built on top of mine.

I now trying to use this one for an family member, as I do not own a unit anymore my self.

https://github.com/wimb0/home-assistant-saj-r5-modbus

santiagoGPNC, 2025-04-23:
@djansen1987  I dont mind taking ownership of the repo, but Im not sure if it makes sense. I have a rebranded inverter and it seems rebrands will remain in the old platform for now (maybe indefenitively), so if I take over it it would be to maintain the old platform, not the new one. But this is quite weird since the integration is called SAJ and it would only work for rebranded (i.e. not-SAJ) equipments.

djansen1987, 2025-04-23:
> @djansen1987  I dont mind taking ownership of the repo, but Im not sure if it makes sense. I have a rebranded inverter and it seems rebrands will remain in the old platform for now (maybe [...]

I understand. You may also create a fork and go from there. Thanks for the help so far!

santiagozky, 2025-04-23:
(back to my personal account...)
I think a fork will make more sense for users in rebranded inverters.

BTW @djansen1987  @marcosturiale . I found  elekeeper does allow for developer accounts. it might be useful for you https://developer.saj-electric.com/fileCenter/index

marcosturiale, 2025-04-23:
After doing some research i found out that the modbus protocol was already available on the device that takes care of sending the data to the cloud. 
I installed the integration you suggested for the r5 inverter, but i didn't get any data. 
After further research i found a specific integration for my h2 s1 inverter (https://github.com/stanus74/home-assistant-saj-h2-modbus) but it is not made available via hacs (or at least not on mine).
I added a custom repository on hacs and installed it: finally i have the data! 
@djansen1987 Check with telnet if you haven't opened port 502 on your inverter's ip, maybe you'll have a nice surprise :)

ThomasNygards, 2025-04-23:
FWIW: I converted the example Java code and C# to Python (mostly using chatGPT)
from 
https://developer.saj-electric.com/fileCenter/index

I can code in python, but I am new to HA components coding.
I have a SAJ R6-17K-T2-32 and am able to test if someone can give me a hand on how to get this tidied up and put into the HACS context. 

Here is the java code converted:

```python
import requests
import json
from collections import OrderedDict
from redis import Redis
import time

class SajApp:
    def __init__(self, enable_encrypt: bool):
        self.enable_encrypt = enable_encrypt
        self.redis_client = Redis()
        self.rest_template = requests.Session()

    def get_access_token(self, app_id: str, app_secret: str) -> str:
        access_token = self.redis_client.get("access_token")
        if access_token:
            return access_token.decode()

        # Fetch new token if not present or expired
        params = {
            "appId": app_id,
            "appSecret": app_secret
        }
        url = "https://pre-developer.saj-electric.com/prod-api/open/api/access_token"
        response = self.rest_template.get(url, params=params)
        result_remote = response.text
        result = json.loads(result_remote)  # Assuming similar to JSONObject.parseObject

        data = result.get("data", {})
        access_token = data.get("access_token")
        access_token_expires = data.get("expires", 28800)  # default 8 hours

        if access_token:
            self.redis_client.setex("access_token", access_token_expires, access_token)
            self.redis_client.set("access_token_expires", access_token_expires)

        return access_token

    def get_device_baseinfo(self):
        app_id = "VH_97d6aiiA"
        app_secret = "qIZ4A7yP6eJLXagfo2lYwdG4aiiXqux2N3Tckt94n3Co97d6hy4SmNMsRF3Pp7jV"
        access_token = self.get_access_token(app_id, app_secret)

        device_sn = "ASS0301G202FE00321"
        params = OrderedDict({
            "appId": app_id,
            "deviceSn": device_sn
        })
        sign = SignUtils.gen_sign(params)

        headers = HttpUtil.gen_http_headers()
        headers["accessToken"] = access_token
        headers["content-language"] = "en_US"

        if self.enable_encrypt:
            aes_key = "2dfc5bda216624b80179fa542bcc342c"  # AesUtils.generate_key()
            client_secret = RsaUtils.public_key_encrypt(aes_key, access_token, 512)
            headers["clientSecret"] = client_secret

            sign = AesUtils.encrypt_ecb(sign, aes_key)

[... 212 lines cut ...]
    api_path = "open/api/device/get/userMode"
    request_url = f"{test_domain}{api_path}?deviceSn={device_sn}"
    result = get_with_headers(request_url, headers)
    print("Read User Mode:", result)

    # Write User Mode (POST)
    user_mode = UserMode(2, device_sn)
    sign_params = {
        "appId": app_id,
        "accUserMode": str(user_mode.accUserMode),
        "deviceSn": user_mode.deviceSn
    }
    client_sign = get_sorted_sign(sign_params)
    headers_post = {
        "accessToken": token_data.data["access_token"],
        "clientSign": client_sign
    }
    api_path = "open/api/device/set/userMode"
    request_url = f"{test_domain}{api_path}"
    result2 = post(request_url, headers_post, user_mode.to_dict())
    print("Write User Mode:", result2)

if __name__ == "__main__":
    main()
```

djansen1987, 2025-04-24:
Thanks for the research, are you able to merge it with the current code and when tested do a pull request? 

Maybe also check if it can be done with less libraries,  seems there are a lot of imports of modules

sematuszewski, 2025-04-24:
I am currently waiting for SAJ to activate my request for approval of the application in the developer account, only then will it be possible to use the api.
As soon as I can get the required approval I will try to implement a test integration.

MicheleMalgeri, 2025-04-29:
I tried to get data scraping web pages and it appears promising. Till now, I accessed to the site and got some data related to production using the library selenium.
I'm patching solar.py in order to maintain the interface as unchanged as possible, but I am not sure about the list and name of parameter the extension need to work.
Finally, I'm not an expert on HA and I need some suggestion in order to test the modification in my installation. Can I simply copy the code in home assistant? How to install the needed libraries?

djansen1987, 2025-04-29:
i think someone here already build a scraper, but i don't know if the code is safe. 

https://github.com/faanskit/ha-esolar/issues/15#issuecomment-2835425600

santiagoGPNC, 2025-04-29:
@djansen1987  if you are moving this integration to the new platform. please make sure to include in the releases notes a warning for people with rebranded drives, since they should NOT upgrade to that new version.

Greetings.

djansen1987, 2025-04-29:
Hi, I have no plans of changing this integration as of now. I do accept pull request if someone is able to adopt the new authentication methode. Which seem to be figured out in the previous link.

Would be great if someone can update this integration with the new urls

teamMOYA, 2025-04-30:
I found an access method similar to the old one that allows data without the need to log in or use developer keys/token. 
My page in this case is: https://eopi.saj-electric.com/bigScreen/rollSinglePlant?viewKey=b03a86e3797fb6117607960ca84df300f94a1593509446824207df7e38c95383fde3e32425f192ad999d4e18e4ac2a68a2b8832604af0263f6bb65c5a007eaeba1b10766078a9b78ac428a79231414bf61b96c0b839231c4a9e4ac61b3b09b76&start=2&lang=en

I don't have enough experience to adapt this information and extract the data. If anyone has knowledge in this area, I would appreciate any help or suggestions.

How i found this:

[image]

[image]

[image]

[image]

[image]

djansen1987, 2025-04-30:
I found that page too, but how would you handle authentication, this page opens using the cookie of the nieuwe login methode right?

djansen1987, 2025-04-30:
Also your screenshots expose personal data of you plant and also an access token which could be used. Probably it is expired already but might be an security issue for your login

teamMOYA, 2025-04-30:
> I found that page too, but how would you handle authentication, this page opens using the cookie of the nieuwe login methode right?

I added this code to the component and It can access the data. I'm trying to adapt it, but I'm having problems due to my lack of knowledge.

```python
            #dmg-i
            _LOGGER.error(f"dmg start")

            # TEST
            PLANT_UID = ''
            DEVICE_SN = ''
            BASE_URL = f"https://eopi.saj-electric.com/bigScreen/getStorePlantPowerInfoForBigScreen?plantuid={PLANT_UID}&devicesn={DEVICE_SN}"
            headers44 = {
                'Accept': 'application/json, text/javascript, */*; q=0.01',
                'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:137.0) Gecko/20100101 Firefox/137.0'
            }

            response44 = await self._session.post(BASE_URL, headers=headers44)

            if response44.status != 200:
                _LOGGER.error(f"{response44.url} dmg returned {response44.status}")
                return

            plantDetails = await response44.json()

            _LOGGER.error(f"PlantDetails: {plantDetails}") #this message shows correct data
            plantDetails.update(plantDetails)
            #dmg-f

```
Obtained data
`2025-05-01 00:07:43.491 ERROR (MainThread) [custom_components.saj_esolar.sensor] PlantDetails: {'storeDevicePower': {'pvPower': 0.0, 'gridPower': 0.0, 'inputOutputPower': 324.0, 'batteryPower': 428.0, 'totalLoadPower': 428.0, 'homeLoadPower': 428.0, 'backupLoadPower': 0.0, 'solarPower': 3.0, 'batCurr': 8.2, 'batEnergyPercent': 60.0, 'runningState': 1, 'isOnline': 1, 'isAlarm': 0, 'mark': 1, 'batCapcity': 100.0, 'batCapcityStr': '100.00Ah', 'hasMeter': True, 'hasBattery': True, 'pvDirection': 0, 'gridDirection': 0, 'batteryDirection': 1, 'outPutDirection': 1, 'hFaultMSG': 0, 'dataTime': 1746021810000, 'updateDate': 1746050600000}, 'type': 2, 'status': 'success'}
`

erelke, 2025-05-01:
I have created a fully functional integration for the new SAJ eleekeper portal. From here, you can install it with the help of hacs: https://github.com/erelke/ha-esolar

djansen1987, 2025-05-02:
> I added this code to the component and It can access the data. I'm trying to adapt it, but I'm having problems due to my lack of knowledge.
>
> ```python
>             #dmg-i [...]

So this works without authentication, or did you perhaps already had a cookie in use with valid login?

konagy, 2025-05-12:
> I have created a fully functional integration for the new SAJ eleekeper portal. From here, you can install it with the help of hacs: https://github.com/erelke/ha-esolar

@erelke , thank for sharing, I already started using it!

odjuvsla, 2025-05-14:
I just installed this integration and seems to work fine for my H2 inverter. Local and no extra devices needed. 

https://github.com/stanus74/home-assistant-saj-h2-modbus